robotology/yarp #1173: yarpmanager cannot launch modules on different machines

giuliavezzani, 2017-04-18:
`yarpmanager` cannot launch modules on different machines (for instance: if you are working on the `icub21`, it cannot launch modules on `icub22` and `icub-cuda`) if the `Log` is enabled on those machines.
I managed to launch the modules on `icub22` and `icub-cuda`, **disabling the their `Log` from the `icub-cluster.py`**.

randaz81, 2017-04-21:
Some recent changes to yarprun have broken the --log option.
@giuliavezzani 
* on icub22 does not work because on that machine the new version of yarprun is running
* on icub21 it works because an old version of yarprun is running (ps aux shows that it has been not restarted since Apr 06th)

@drdanz 
Maybe https://github.com/robotology/yarp/commit/3c1882dfb8a217b4e3a83962b49aeba417c9aee1 ?

randaz81, 2017-04-21:
The error message displayed by yarprun:
```
ABORTED: server=/icub22 alias=test cmd=stdout
Can't execute stdout because Operation not supported
```
In the previous version, yarprun was displaying the **name of the executable to run** instead of **stdout**

Tobias-Fischer, 2017-04-26:
Hi,
Do you already have an update on this? We were just running into this issue. Anything I can help with to resolve this?

Best, Tobias

claudiofantacci, 2017-04-26:
Hi @Tobias-Fischer, afaik there is not a fix yet, but is on the go 😄 
Thanks for the patience!

drdanz, 2017-05-03:
After some debugging, I think there was some change on debian/ubuntu that break it

my test was:
```bash
yarp server
```
```bash
yarprun --server /foo --log
```
```bash
yarprun --on /foo --as bar --cmd /bin/true
```

Result:
* Works on windows 10
* Broken on 2 different debian testing (one recently updated and one updated a few weeks ago)
* Broken on ubuntu xenial
* Works on ubuntu trusty

I bisected it and apparently the first broken commit was the one where the --log option was added

Can anyone test it on more (or even the same) operating systems?

francesco-romano, 2017-05-03:
macOS 10.12.4

```
makas-pro:build makaveli$ yarprun --server /foo --log
STARTED: server=/foo alias=bar cmd=stdout pid=65483
ABORTED: server=/foo alias=bar cmd=stdout
Can't execute stdout because Operation not supported
ABORTED: server=/foo alias=bar cmd=stdout
Can't execute stdout because Operation not supported
ABORTED: server=/foo alias=bar cmd=/bin/true
Can't execute command because No such file or directory

```

```
makas-pro:build makaveli$ yarprun --on /foo --as bar --cmd /bin/true
yarp: Port /tmp/port/1 active at tcp://10.255.36.7:10003/
yarp: Sending output from /tmp/port/1 to /foo using tcp
yarp: Removing output from /tmp/port/1 to /foo
RESPONSE:
=========
-1
ABORTED: server=/foo alias=bar cmd=stdout
Can't execute stdout because Operation not supported
ABORTED: server=/foo alias=bar cmd=/bin/true
Can't execute command because No such file or directory
```

pattacini, 2017-05-03:
Great debugging @drdanz 👍 
However, I can certainly tell that the `--log` was running ok on `xenial` in the past, just because we have introduced `xenial` in our system since quite long by now.

francesco-romano, 2017-05-03:
If I may add, on macOS using `master` branch everything works perfectly.

```
makas-pro:build makaveli$ yarprun --server /foo --log
STARTED: server=/foo alias=bar cmd=stdout pid=67235
STARTED: server=/foo alias=bar cmd=/bin/ls pid=67236
```

```
yarprun --on /foo --as bar --cmd /bin/ls
yarp: Port /tmp/port/1 active at tcp://10.255.36.7:10003/
yarp: Sending output from /tmp/port/1 to /foo using tcp
yarp: Removing output from /tmp/port/1 to /foo
RESPONSE:
=========
67236
STARTED: server=/foo alias=bar cmd=/bin/ls pid=67236
```

claudiofantacci, 2017-05-03:
It is broken also on `Linux 4.6.0-1-amd64 #1 SMP Debian 4.6.4-1 (2016-07-18) x86_64 GNU/Linux` running on my laptop.

drdanz, 2017-05-03:
Ok, found the problem, I couldn't find it on my systems since for some reason I was building with `SKIP_ACE` on both of them, and I had issues with the PATH variable...
Anyway, it should be fixed now...

drdanz, 2017-05-03:
Please everyone test and report...

closed 2017-05-03 by drdanz

francesco-romano, 2017-05-03:
@drdanz tested on macOS and the issue is solved.

traversaro, 2017-05-04:
@gabrielenava is testing it now on iCubGenova04 , I will update this issue.

amaroyo, 2017-05-04:
@drdanz Updated and tested on iCubGenova03. Everything running smoothly :) Thanks!

commit b951c5c3 (2017-05-03): OS: do not call ACE_OS:::execlp that is not implemented

Fixes #1173