robotology/yarp #1178: Unknown CMake command "swig_add_library" error, while compiling java bindings

AL458, 2017-04-20:
Hi all,

I cannot compile yarp java bindings, as cmake apparently is not able to find the swig_add_library command. The ccmake GUI reports the following error:

CMake Error at CMakeLists.txt:145 (swig_add_library):
Unknown CMake command "swig_add_library".

I followed the instructions on http://www.yarp.it/yarp_swig.html#yarp_swig_configure , no way to make it work.

all the option settings look correct (including "SWIG_DIR /usr/share/swig3.0" and "SWIG_EXECUTABLE /usr/bin/swig3.0")

OS: ubuntu 16.04
cmake version: 3.5.1
swig version: 3.0.8
yarp version: 2.3.68

Thanks in advance!

traversaro, 2017-04-24:
Hi @AL458 , 
   did you follow the instructions " Configure bindings while compiling YARP" or " Configure bindings to build separately from YARP" ? The first option is the most tested and the one that should work fine. The second should be removed from the documentation because it is highly confusing.

AL458, 2017-04-26:
Hi @traversaro ,

thanks for your answer. I tried the second option, as I decided to build the bindings after having installed YARP. Anyway, I solved the problem. It was my CMake version (3.5). Once I installed the latest (3.8), everything worked fine.

Thanks again!

closed 2017-04-26 by AL458