robotology/yarp #1179: iCub cluster - yarp run won't work

pattacini, 2017-04-21:
@amaroyo commented on [Fri Apr 21 2017](https://github.com/robotology/icub-main/issues/440)

Hello, 

Just to set up the background: I am on the branch devel in yarp and icub-main. I just have done an update and fresh installation (removing build folders) on pc104 and the main terminal of the setup (icub14). 

I have the following problem:

I am able to run the icub-cluster.py, start the server and run it on different nodes (let's just stick to pc104 and icub14). It runs happily and there are no apparent problems. However, when I try to run a module from yarpmanager on icub14 or pc104 I have the following error:

```cannot run yarpview on icub14: cannot cask /icub14 to run yarpview (Undefined message)```

Just a note, all the modules run satisfactorily through the shell. 

Next step I did is to run the module using the following command in the shell:

```yarp run --on /icub14 --as yView --cmd yarpview``` 

and the error output is this one:
```
yarp: Port /tmp/port/1 active at tcp://10.0.0.134:10004/
yarp: Sending output from /tmp/port/1 to /icub14 using tcp
yarp: Removing output from /tmp/port/1 to /icub14
RESPONSE:
=========
-1
ABORTED: server=/icub14 alias=yView cmd=stdout
Can't execute stdout because Operation not supported
```

Which I track down to the following file in yarp src [**```Run.cpp```**](https://github.com/robotology/yarp/blob/master/src/libYARP_OS/src/Run.cpp#L1670) (line in the link).

**However,** the "funny" part is that I stop the nodes of pc104 and icub14 on the icub-claster.py, 
I run the following command through the sell to make them yarp aware:

```yarp run --server /icub14```

and I can start the process even through yarpmanager, and it works!!

So I really don't know what's going on here. That's why I did again a fresh installation of yarp and icub-main.

P.S.: there are no error / warning messages in the shell in which icub-cluster.py is running.

pattacini, 2017-04-21:
It might be related to #1173

randaz81, 2017-04-21:
@pattacini indeed, it is exactly the same issue!

pattacini, 2017-04-21:
Then, let's close it in favor of the original one.

closed 2017-04-21 by pattacini