robotology/yarp #1185: Installing yarp on OSX with Homebrew

ClementLeBihan, 2017-04-22:
Hi, I'm trying to install yarp on OSX Sierra, but I have these message : 

```
brew tap robotology/cask
==> Tapping robotology/cask
Cloning into '/usr/local/Homebrew/Library/Taps/robotology/homebrew-cask'...
remote: Counting objects: 6, done.
remote: Compressing objects: 100% (6/6), done.
remote: Total 6 (delta 0), reused 4 (delta 0), pack-reused 0
Unpacking objects: 100% (6/6), done.
Checking connectivity... done.
Error: Cannot tap robotology/cask: invalid syntax in tap!
```
I think that's because cask is part of home-brew now. However, I can't install yarp :
```
brew install yarp
Error: No available formula with the name "yarp" 
==> Searching for a previously deleted formula...
Error: No previously deleted formula found.
==> Searching for similarly named formulae...
Error: No similarly named formulae found.
==> Searching taps...
Error: No formulae found in taps.
```

Any ideas ?

claudiofantacci, 2017-04-23:
Hi @ClementLeBihan ,

I'm on macOS Sierra and have no problem with it. Here is my output:
```bash
Claudio$ brew tap robotology/cask
==> Tapping robotology/cask
Cloning into '/usr/local/Homebrew/Library/Taps/robotology/homebrew-cask'...
remote: Counting objects: 6, done.
remote: Compressing objects: 100% (6/6), done.
remote: Total 6 (delta 0), reused 4 (delta 0), pack-reused 0
Unpacking objects: 100% (6/6), done.
Tapped 3 formulae (32 files, 30KB)
```

Did you `brew update`?
Here is my version:
```bash
Claudio$ brew --version 
Homebrew 1.1.13-93-gf3dc06a4e
Homebrew/homebrew-core (git revision 2db29; last commit 2017-04-21)
```

ClementLeBihan, 2017-04-23:
Hi claudio !

Unfortunately I'm under the 1.1.13-93 : 
```
brew --version
Homebrew 1.1.13-79-g5bf0745
Homebrew/homebrew-core (git revision a92c0; last commit 2017-04-23)
```

But my brew update output is : 
```
brew update
Already up-to-date.
```

Do you have any idea ?

claudiofantacci, 2017-04-23:
That is strange, did you try `brew tap homebrew/cask` again?

Otherwise, something else went wrong and we need to figure it out.
Try issuing
```bash
cd $(brew --repo)
git status
```

ClementLeBihan, 2017-04-23:
> brew tap homebrew/cask :

```
brew tap homebrew/cask
==> Tapping homebrew/cask
Cloning into '/usr/local/Homebrew/Library/Taps/homebrew/homebrew-cask'...
remote: Repository not found.
fatal: repository 'https://github.com/Homebrew/homebrew-cask/' not found
Error: Failure while executing: git clone https://github.com/Homebrew/homebrew-cask /usr/local/Homebrew/Library/Taps/homebrew/homebrew-cask --depth=1
```

> cd $(brew --repo)
> git status

```
git status
On branch master
Your branch is up-to-date with 'origin/master'.
nothing to commit, working directory clean
```

ClementLeBihan, 2017-04-23:
#https://github.com/caskroom/homebrew-cask/issues/32840#issuecomment-296434059

claudiofantacci, 2017-04-23:
You probably have a bugged version of `brew`, but you are unable to update it.

Try with `brew update --force`.
Also please do a `brew doctor` and post here the output.

ClementLeBihan, 2017-04-23:
God thank you ! `brew update --force` output Already up-to-date. but I can now tap robotology/cask and install yarp ! Thank you so much @claudiofantacci !

closed 2017-04-23 by ClementLeBihan

claudiofantacci, 2017-04-23:
Glad to be of help!