robotology/yarp #1187: CMake policy CMP0054 warnings with cmake 3.8

claudiofantacci, 2017-04-24:
Updating to CMake 3.8 and compiling YARP gives a flood of the following [CMP0054](https://cmake.org/cmake/help/v3.8/policy/CMP0054.html) __warnings__:
```cmake
CMake Warning (dev) at cmake/ycm-0.3.1/modules/GetAllCMakeProperties.cmake:37 (if):
  Policy CMP0054 is not set: Only interpret if() arguments as variables or
  keywords when unquoted.  Run "cmake --help-policy CMP0054" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.
  [...]
```

```cmake
CMake Warning (dev) at cmake/YarpOSXUtilities.cmake:42 (if):
  Policy CMP0054 is not set: Only interpret if() arguments as variables or
  keywords when unquoted.  Run "cmake --help-policy CMP0054" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.
  [...]
```

The warning is generated by the two files `YarpOSXUtilities.cmake` and `GetAllCMakeProperties.cmake` (the latter is part of ycm 3.1 and copied in YARP devel branch) only.

_To make the long story short_,
the warnings are generated by the following line `if("${_prop_name}" STREQUAL "TYPE")`, and similar ones in end-if control flows. With the new policy, from CMake 3.1, variables between double quotes are expanded only once.
I checked `_prop_name` and it seems there is no need for the double quotes. Further, I tried compiling YARP with this changes and seems to be doing just fine. Thus it seems that `_prop_name` is not itself a container of other variables to be dereferenced.
What do you think about it @drdanz ?

I can open a PR to fix this for `YarpOSXUtilities.cmake`, but, as far as `GetAllCMakeProperties.cmake` is concerned, it is not clear to me whether I have to open a PR on `ycm` or on `YARP` and, more importantly, whether we can replace `GetAllCMakeProperties.cmake` with [GetAllCMakeProperties](https://cmake.org/cmake/help/v3.8/command/cmake_parse_arguments.html#command:cmake_parse_arguments), which is shipped with CMake since version 3.0, or not.

claudiofantacci, 2017-05-10:
@drdanz let me know whether the fix I have in my `YARP` fork is ok for a PR or it is better to do a PR on `YCM` and only then update the affected files in `YARP`.

drdanz, 2017-05-16:
@claudiofantacci The right place is in YCM, otherwise we will have to sync it backwards...

claudiofantacci, 2017-05-16:
`YarpOSXUtilities.cmak` is `YARP` only.
Furthermore, I cant' really check all the files in YCM but for `GetAllCMakeProperties .cmake` which is used by `YARP`.
I think I will just modify these two files.

One more thing: I think that the fix I proposed earlier is ok, but maybe it will be clearer to just add `cmake_policy(SET CMP0054 NEW)`. @drdanz what do you suggest?

drdanz, 2017-06-05:
Fixed by #1225

PR #1225 (2017-06-05): Fix CMake policy CMP0054
This commit, if applied, enforces the new [CMP0054](https://cmake.org/cmake/help/v3.8/policy/CMP0054.html) behavior, if available in the current CMake version.

Fixes #1187 
Integrates [robotology/ycm#116](https://github.com/robotology/ycm/issues/116)
commit 09a5bf67 (2017-06-05): Fix CMake policy CMP0054 warnings for CMake >3.8.0

Fixes #1187
closed 2017-06-05 by drdanz