robotology/yarp #1190: Cannot make connection to 'xmlrpc

pedrombmachado, 2017-05-01:
Dear All,
  I am trying to configure YARP to work ROS but I am getting the error:
Cannot connect to ROS subscriber http://cncr-1:41205/
Cannot make connection to 'xmlrpc://cncr-1:41205'

[image]

I have compiled yarp from the sources and I followed the instructions described [here](http://wiki.icub.org/wiki/YARP_ROS_Interoperation_Past_and_Future_But_Not_Now) and [here](http://www.yarp.it/yarp_with_ros_nameservers.html). 

Please find below CMakeCache.txt used to compile yarp:
[CMakeCache.txt](https://github.com/robotology/yarp/files/967562/CMakeCache.txt)

pedro@cncr-1:~$ yarp version
YARP version 2.3.68+230-20170426.1+git4c562a6+dirty

ros_comm version 1.12.7

Linux cncr-1 4.8.0-49-generic #52~16.04.1-Ubuntu SMP Thu Apr 20 10:55:59 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

plinioMoreno, 2017-05-05:
@pedrombmachado, from the screenshot it looks like the variable ROS_MASTER_URI is not defined properly. roscore is reporting `ROS_MASTER_URI=http://cncr-1:11311` and `yarpdetect` is reporting `ROS_MASTER_URI=http://localhost:11311`. Both should report the same value

drdanz, 2017-05-05:
Hello @pedrombmachado 
Can you post the content of the `yarp.conf` file in "~/.config/yarp`?

pedrombmachado, 2017-05-05:
Hi @drdanz ,
  It says
`$ cat ~/.config/yarp/yarp.conf 
10.42.0.22 10000 yarp`

pedrombmachado, 2017-05-05:
Hi @plinioMoreno ,
  Done.

pedrombmachado, 2017-05-05:
Still not working :(

drdanz, 2017-05-16:
@pedrombmachado the configuration is for a normal yarp server, but if I understand correctly you are trying to use ros without the yarp server, correct?
Can you try 'yarp detect --ros --write'?

pedrombmachado, 2017-05-16:
Hi @drdanz ,
  I was using the the yarpserver and roscore. But never mind because I worked around and I simplified. I have YARP-Python environment and I am able to use it with opencv. Therefore I am going to close this issue.
Many thanks,
Pedro

closed 2017-05-16 by pedrombmachado

drdanz, 2017-05-17:
@pedrombmachado Sorry if we couldn't help on this specific issue, I'm glad you managed to work around it.