robotology/yarp #1197: 'yarp name get /port accepts' does NOT return anything

aikolina, 2017-05-05:
Hello, 

I've just started to follow the tutorial on YARP ([Getting started with YARP ports](http://www.yarp.it/note_ports.html)) and I had a little problem while testing the command `yarp name get /port accepts` on Ubuntu 16.04.

According to the tutorial, the result of the command should be the following one:
```
[you type] yarp name get /summer accepts
[response] port /summer property accepts = tcp text text_ack udp mcast shmem
           *** end of message
```
On the contrary, in my case I get nothing:
```
adinale@iitrbcslap125:~$ yarp name get /summer accepts
port /summer property accepts =
*** end of message
```
But apparently, other people who has tested this command got at least a partial results like:
```
port /summer property accepts = udp mcast
*** end of message
```
So I have few questions regarding this issue:
- Is this a common misbehavior?
- Do I need some specific setup to make this command work?
- Is there an updated version of this command that I should use?

Thanks!

**NOTE:** I would like to underline that the ports that I created before reaching this critical point of the tutorial work perfectly!

drdanz, 2017-05-05:
@aikolina Are you using `master` or `devel`

aikolina, 2017-05-05:
Actually, I installed YARP from the binaries. So probably it's not the `devel` branch.

traversaro, 2017-05-05:
On the latest `devel` on Ubuntu 16.04 I also get the same empty list as result of the `accept` command, opening two ports using `yarp read` and `yarp write` .

drdanz, 2017-06-01:
After some more debugging, I found out that everything seems to be working, excepts for the part that inserts the property "accepts" into yarpserver database.
Executing ``yarp name set /summer accepts tcp text text_ack udp mcast shmem`` sets this property, and therefore any following ``yarp name get /summer accepts`` returns the expected output.

Somewhere hidden inside YARP there should have been some place where this `set` command was executed.

drdanz, 2017-06-01:
@Nicogene found out that in `NameClient.cpp` this part was commented out.

The responsible commit seems to be 02f9b1983614ef08a2ada598a38a357e68723731

Nicogene, 2017-06-01:
Fixed by #1249

drdanz, 2017-06-01:
I was right, it was already there in YARP 2.3.20... actually it was there in since v2.3.7 :+1:

PR #1249 (2017-06-01): NameClient: fix yarp name /port accepts
This PR Fix #1197.
Now it actually check the available carriers in the machine.

Please review code.
PR #1237 (2017-06-22): yarpmanager: add the scan of available carriers before performing connect and disconnect
This PR introduces:
- [x]  scan among the available carriers when you click connect and disconnect, if the specified carrier is not available a proper error message will be written on log.
- [x] the column *carrier* in the connection list has been substituted with a combobox with the available carriers
- [x] add modifiers column in the connection list
- [x] Fix #1197 
- [x] add automatic ports/resources refresh after `run/stop/kill`
- [x] test on `iCubGenova01` setup

Please review code.
closed 2017-06-01 by drdanz