robotology/yarp #1198: cannot use CMake 3.8.1 (?)

randaz81, 2017-05-05:
After upgrading CMake to 3.8.1 I get the following error:

```
CMake Error at src/libYARP_init/CMakeLists.txt:15 (source_group):
  source_group ROOT: C:/software/yarp/src/libYARP_init is not a prefix of
  file: C:/Program Files/CMake/bin/src/CustomInit.cpp

CMake Error at src/yarpview/plugin/CMakeLists.txt:30 (source_group):
  source_group ROOT: C:/software/yarp/src/yarpview/plugin is not a prefix of
  file: C:/Program Files/CMake/bin/qtyarpview_plugin.cpp

CMake Error at src/yarpview/plugin/CMakeLists.txt:33 (source_group):
  source_group ROOT: C:/software/yarp/src/yarpview/plugin is not a prefix of
  file: C:/Program Files/CMake/bin/qtyarpview.h

CMake Error at src/yarpview/src/CMakeLists.txt:17 (source_group):
  source_group ROOT: C:/software/yarp/src/yarpview/src is not a prefix of
  file: C:/Program Files/CMake/bin/main.cpp

CMake Error at src/yarpview/src/CMakeLists.txt:20 (source_group):
  source_group ROOT: C:/software/yarp/src/yarpview/src is not a prefix of
  file: C:/Program
  Files/CMake/bin/qtquick2applicationviewer/qtquick2applicationviewer.h

CMake Error at src/yarpscope/plugin/CMakeLists.txt:33 (source_group):
  source_group ROOT: C:/software/yarp/src/yarpscope/plugin is not a prefix of
  file: C:/Program Files/CMake/bin/portreader.cpp

CMake Error at src/yarpscope/plugin/CMakeLists.txt:36 (source_group):
  source_group ROOT: C:/software/yarp/src/yarpscope/plugin is not a prefix of
  file: C:/Program Files/CMake/bin/portreader.h

CMake Error at src/yarpscope/src/CMakeLists.txt:17 (source_group):
  source_group ROOT: C:/software/yarp/src/yarpscope/src is not a prefix of
  file: C:/Program Files/CMake/bin/main.cpp

CMake Error at src/yarpscope/src/CMakeLists.txt:20 (source_group):
  source_group ROOT: C:/software/yarp/src/yarpscope/src is not a prefix of
  file: C:/Program
  Files/CMake/bin/qtquick2applicationviewer/qtquick2applicationviewer.h

CMake Error at src/yarpmanager/CMakeLists.txt:114 (source_group):
  source_group ROOT: C:/software/yarp/src/yarpmanager is not a prefix of
  file: C:/Program Files/CMake/bin/src-manager/aboutdlg.cpp

CMake Error at src/yarpmanager/CMakeLists.txt:118 (source_group):
  source_group ROOT: C:/software/yarp/src/yarpmanager is not a prefix of
  file: C:/Program Files/CMake/bin/src-manager/aboutdlg.h

CMake Error at src/yarpdataplayer/CMakeLists.txt:66 (source_group):
  source_group ROOT: C:/software/yarp/src/yarpdataplayer is not a prefix of
  file: C:/Program Files/CMake/bin/src/aboutdlg.cpp

CMake Error at src/yarpdataplayer/CMakeLists.txt:69 (source_group):
  source_group ROOT: C:/software/yarp/src/yarpdataplayer is not a prefix of
  file: C:/Program Files/CMake/bin/include/aboutdlg.h

CMake Error at src/yarpmotorgui/CMakeLists.txt:52 (source_group):
  source_group ROOT: C:/software/yarp/src/yarpmotorgui is not a prefix of
  file: C:/Program Files/CMake/bin/main.cpp

CMake Error at src/yarpmotorgui/CMakeLists.txt:55 (source_group):
  source_group ROOT: C:/software/yarp/src/yarpmotorgui is not a prefix of
  file: C:/Program Files/CMake/bin/flowlayout.h

CMake Error at src/yarpbatterygui/CMakeLists.txt:31 (source_group):
  source_group ROOT: C:/software/yarp/src/yarpbatterygui is not a prefix of
  file: C:/Program Files/CMake/bin/main.cpp

CMake Error at src/yarpbatterygui/CMakeLists.txt:34 (source_group):
  source_group ROOT: C:/software/yarp/src/yarpbatterygui is not a prefix of
  file: C:/Program Files/CMake/bin/display.h

CMake Error at src/yarplaserscannergui/CMakeLists.txt:32 (source_group):
  source_group ROOT: C:/software/yarp/src/yarplaserscannergui is not a prefix
  of file: C:/Program Files/CMake/bin/main.cpp

CMake Error at src/libYARP_rtf/CMakeLists.txt:18 (source_group):
  source_group ROOT: C:/software/yarp/src/libYARP_rtf is not a prefix of
  file: C:/Program Files/CMake/bin/src/JointsPosMotion.cpp

CMake Error at src/libYARP_rtf/CMakeLists.txt:21 (source_group):
  source_group ROOT: C:/software/yarp/src/libYARP_rtf is not a prefix of
  file: C:/Program Files/CMake/bin/include/yarp/rtf/JointsPosMotion.h
```

randaz81, 2017-05-05:
I switched back to CMake 3.7.2...

drdanz, 2017-05-05:
I think the issue is in CMake 3.8.1 (apparently it was working on 3.8.0 according to @pattacini) but further testing is required.

pattacini, 2017-05-05:
@drdanz @randaz81 I can confirm it's working with CMake 3.8.0

traversaro, 2017-05-05:
Probably related to CMake issue https://gitlab.kitware.com/cmake/cmake/issues/16807 and the relative MR https://gitlab.kitware.com/cmake/cmake/merge_requests/731 .

traversaro, 2017-05-06:
Note that CMake 3.8.1 was released in homebrew, but I don't know if this issue affects also macOS.

traversaro, 2017-05-06:
I must state the obvious, but this problem is related to https://github.com/robotology/yarp/commit/df92d9d0eb2a3171e6eee58f44f34d4c43171df6 and so the `master` branch works fine.

traversaro, 2017-05-06:
A workaround (to force the old behaviour) is to comment out the `return()` in https://github.com/robotology/yarp/blob/devel/cmake/YarpSourceGroup.cmake#L9 .

drdanz, 2017-05-16:
This is currently fixed with a workaround, we should work upstream to have it working in CMake 3.8.2 https://gitlab.kitware.com/cmake/cmake/issues/16876

closed 2017-05-16 by drdanz