robotology/yarp #1212: Segmentation fault in yarp::dev::ControlBoardSubControlBoardAxesDecomposition

diegoferigo, 2017-05-12:
I got the following segfault while trying to run `yarprobotinterface` for a Gazebo 7 simulation:

[image]

Steps for reproducing the segfault:

```sh
yarpserver &
gazebo &
YARP_ROBOT_NAME="icubGazeboSim" yarprobotinterface --config launch-wholebodydynamics.xml
```

Eventually, I'm not sure if the `yarprobotinterface` output could be relevant:
```
TcpConnector::connect fail: Connection refused
Connect [handle=48] at 172.18.0.4:16679
yarp: SocketTwoWayStream::updateAddresses failed getting local address
yarp: Removing output from /icubSim/right_arm/stateExt:o to /wholeBodyDynamics/right_arm/stateExt:i
 * query "/wholeBodyDynamics/r_leg_ft_sensor"
 * query "/wholeBodyDynamics/left_leg/stateExt:i"
TcpConnector::connect fail: Connection refused
Connect [handle=56] at 172.18.0.4:21031
yarp: SocketTwoWayStream::updateAddresses failed getting local address
yarp: Removing output from /icubSim/right_leg/analog:o to /wholeBodyDynamics/r_leg_ft_sensor
TcpConnector::connect fail: Connection refused
Connect [handle=51] at 172.18.0.4:17447
yarp: SocketTwoWayStream::updateAddresses failed getting local address
yarp: Removing output from /icubSim/left_leg/stateExt:o to /wholeBodyDynamics/left_leg/stateExt:i
 * query "/wholeBodyDynamics/right_leg/stateExt:i"
TcpConnector::connect fail: Connection refused
Connect [handle=51] at 172.18.0.4:18215
yarp: SocketTwoWayStream::updateAddresses failed getting local address
yarp: Removing output from /icubSim/right_leg/stateExt:o to /wholeBodyDynamics/right_leg/stateExt:i
 * query "/wholeBodyDynamics/l_arm_ft_sensor"
TcpConnector::connect fail: Connection refused
Connect [handle=51] at 172.18.0.4:19495
yarp: SocketTwoWayStream::updateAddresses failed getting local address
yarp: Removing output from /icubSim/left_arm/analog:o to /wholeBodyDynamics/l_arm_ft_sensor
 * query "/wholeBodyDynamics/l_foot_ft_sensor"
TcpConnector::connect fail: Connection refused
Connect [handle=51] at 172.18.0.4:21543
yarp: SocketTwoWayStream::updateAddresses failed getting local address
yarp: Removing output from /icubSim/left_foot/analog:o to /wholeBodyDynamics/l_foot_ft_sensor
 * query "/wholeBodyDynamics/head/stateExt:i"
TcpConnector::connect fail: Connection refused
Connect [handle=51] at 172.18.0.4:18983
yarp: SocketTwoWayStream::updateAddresses failed getting local address
yarp: Removing output from /icubSim/head/stateExt:o to /wholeBodyDynamics/head/stateExt:i
 * query "/wholeBodyDynamics/r_arm_ft_sensor"
 * query "/wholeBodyDynamics/torso/stateExt:i"
TcpConnector::connect fail: Connection refused
Connect [handle=51] at 172.18.0.4:20007
yarp: SocketTwoWayStream::updateAddresses failed getting local address
yarp: Removing output from /icubSim/right_arm/analog:o to /wholeBodyDynamics/r_arm_ft_sensor
TcpConnector::connect fail: Connection refused
Connect [handle=51] at 172.18.0.4:15143
yarp: SocketTwoWayStream::updateAddresses failed getting local address
yarp: Removing output from /icubSim/torso/stateExt:o to /wholeBodyDynamics/torso/stateExt:i
 * query "/wholeBodyDynamics/r_foot_ft_sensor"
 * query "/wholeBodyDynamics/imu"
TcpConnector::connect fail: Connection refused
Connect [handle=51] at 172.18.0.4:19239
yarp: SocketTwoWayStream::updateAddresses failed getting local address
yarp: Removing output from /icubSim/inertial to /wholeBodyDynamics/imu
TcpConnector::connect fail: Connection refused
Connect [handle=56] at 172.18.0.4:22055
yarp: SocketTwoWayStream::updateAddresses failed getting local address
yarp: Removing output from /icubSim/right_foot/analog:o to /wholeBodyDynamics/r_foot_ft_sensor
 * query "/wholeBodyDynamics/l_leg_ft_sensor"
 * query "/wholeBodyDynamics/left_arm/stateExt:i"
TcpConnector::connect fail: Connection refused
Connect [handle=48] at 172.18.0.4:20519
yarp: SocketTwoWayStream::updateAddresses failed getting local address
yarp: Removing output from /icubSim/left_leg/analog:o to /wholeBodyDynamics/l_leg_ft_sensor
TcpConnector::connect fail: Connection refused
Connect [handle=48] at 172.18.0.4:15911
yarp: SocketTwoWayStream::updateAddresses failed getting local address
yarp: Removing output from /icubSim/left_arm/stateExt:o to /wholeBodyDynamics/left_arm/stateExt:i
```

diegoferigo, 2017-05-12:
I just realized this is not `yarp` code but [robotology-playground/human-dynamics-estimation](https://github.com/robotology-playground/human-dynamics-estimation) code. You can safely remove this issue, I open it again there.

traversaro, 2017-05-12:
Interestingly, the `getControlModes` method should be tested in https://github.com/robotology/yarp/blob/359cec516bfe85fdee8bebda64835ff0eec26a6e/tests/libYARP_dev/ControlBoardRemapperTest.cpp#L120 and the same test run under valgrind memcheck in https://github.com/robotology/yarp/blob/359cec516bfe85fdee8bebda64835ff0eec26a6e/tests/CMakeLists.txt#L136 .

diegoferigo, 2017-05-12:
After some debugging, the culprit seems to be `full[j]` ([Line 279](https://github.com/robotology/yarp/blob/729eca738ed1b73bf4fb71c8e111892a87cdd654/src/libYARP_dev/src/devices/ControlBoardRemapper/ControlBoardRemapperHelpers.cpp#L279)) that is a `nullptr`.

diegoferigo, 2017-05-12:
With the help of @traversaro, we figured out that it probably was a linking problem. From the valgrind trace, it can be seen that `yarp::dev::WholeBodyDynamicsDevice::publishGravityCompensation()` calls `yarp::dev::ControlBoardRemapper::getControlModes(int*)`, but browsing the code, this latter function is not actually called.

The source situation was:

* `yarp`
* `codyco-superbuild` <-- `codyco-modules` <-- `WholeBodyDynamicsDevice`

The installed `codyco-modules` was built against `yarp:master`, and this issue showed up after switching to `yarp:devel`. I didn't rebuilt `codyco-modules`, and this was the problem. Probably methods in `ControlBoardRemapper` have been added / removed, and the offset pointing to `getControlMode()` (right function to call) finished pointing to `getControlModes()`, generating the segfault.

closed 2017-05-12 by diegoferigo