robotology/yarp #1215: Discussion about real time thread priorities

diegoferigo, 2017-05-16:
The current status of real time thread support in `yarp` is still in experimental status and hence undocumented. This could be a useful feature for critic threads that run into `icub-head`, and could potentially mitigate situations like https://github.com/robotology/icub-support/issues/449.

Before starting, Some bits of history, at least what I found so far.

Some year ago `yarp` gained the experimental support of tweaking

DanielePucci, 2017-05-16:
@lornat75 @marcoaccame @randaz81 @pattacini @traversaro @francesco-romano

marcoaccame, 2017-05-17:
Hi all, 
I have some more info about previous history of real time threads in robotInterface (around 2014).
Let me search the related redmine issues. I will soon post a brief on that.

marcoaccame, 2017-05-17:
**Here is some more history.**

1. There was a time (about end of 2014) when robotInterface threads had SCHED_FIFO scheduling policy and ad-hoc RT priorities. This mechanism was introduced to solve some problems we experienced such as sporadic packet loss.  
In particular: all robotInterface threads had scheduling policy SCHED_FIFO. All had priority 33, apart from thread EthSender (49) and EthReceiver (48). See `Note-1` for my comments on these priorities. 
I recall that @apaikan and I worked with RT another time, in 2015, to reduce the CPU load of old PC104.
See `Appendix` for details of both activities.
2. The RT mode was enabled by means of the ICUB_REALTIME_EXPERIMENTAL CMake variable which defines the value of macro ICUB_USE_REALTIME_LINUX which adds to the code some Linux system calls. At the time we did not have YARP support and we had to use Linux system calls, making the system non-portable.
4. Some time after YARP gained RT support but the code under macro ICUB_USE_REALTIME_LINUX was not changed to use the new APIs.
3. For several reasons, the yarprobotinterface threads don't have this RT feature anymore. Reasons are:
   - The default value of the ICUB_REALTIME_EXPERIMENTAL variable has always been 0 and it never became 1. 
   - Documentation in http://wiki.icub.org/wiki/Compilation_on_the_pc104 does not mention the use of 
       this variable, hence new installations don't have RT mode turned on.
   - When the main.cpp of robotInterface moved to YARP, the ICUB_REALTIME_EXPERIMENTAL did not move as well with the result that the code under macro ICUB_USE_REALTIME_LINUX is never compiled.
5. In some robots we now have a more powerful processor (new COM Express vs old PC104) with a local hard disk. These features much improve computation capabilities and system responsiveness.

Now, I recall that in the time of old PC104 the RT mode was required to mitigate packet loss. So, why now ICUB_USE_REALTIME_LINUX is off and we dont see the problems of years ago? 

I think the reason is in point (5): now the system has much more computation capabilities and 
the problem does not (statistically?) shows. 

However, I think that the improved computation capabilities are not enough to guarantee a safe system.
 
**And here is a proposal for a way forward:**
1. We should at least revert back to RT mode as tested some time ago. 
2. After, I agree with @diegoferigo, we should define / experiment a proper assignment of priorities to the threads running inside icub-head. In `Note-2` are some thoughts on priorities.

---
### Notes

`Note-1`
I don't recall the reason of numbers 33, 48 and 49. I remember only that originally they were 33, 49 and 49 and later we moved one 49 into 48. Maybe @apaikan remembers why.

`Note-2`
If I should define priorities now I would chose values in the range [50, 99], so that they sit between the system IRQs which maintain the highest priorities and the remain common user threads. Moreover, I would assign highest priority to the EthSender (it normally executes for shorter time and we really want to deliver commands to the motors ASAP). Then, as second, I would put the EthReceiver (because it parses UDP packets, unlocks blocking requests and fills all data coming from the ETH boards inside the devices -  embObjStrain, embObjMotionControl, etc. -, data which is normally read at 10 ms rate).  Finally, as third I would put all other threads in yarprobotinterface (because I assume they just deal with port communication ... but see my `Question-1`).

---
### Questions

`Question-1` What are the other threads in yarprobotinteface? Those associated to the wrappers?  

---

### Appendix

Here are details of past activities reported in redmine.

#### Issue about Packet loss in PC104

In https://redmine.robotology.eu/issues/87 it is fist mentioned that in Oct 2014 @apaikan introduced a variable to toggle on /off `realtime kernel with FIFO scheduling and suitable priorities in threads of robotInterface`.

I tested that feature for the sake of solving the problem of packet loss in PC104.

For those of you who don't have access to redmine, here are some excerpts from the issue:

`Marco Accame 13 Oct 2014 07:31 PM`
```

The result of these tests is that the packet loss happens in the PC104 and is due to 
incorrect scheduling of the receiving thread and to small size of the buffer size. 
By applying suitable small changes the existing kernel setup and in the code of 
robotInterface, the problem seems to be solved.

The brief of all the changes which solve the problem is in the following.

In system:
1. Use of a realtime kernel with FIFO scheduling 
2. Increased max size of UDP receiving buffer to 8MB: added line "net.core.rmem_max=8388608" 
   in file /etc/sysctl.conf
3. Given to icub user the right to run high priority threads: added the two lines 
   "icub soft rtprio 99" and "icub hard rtprio 99" in file /etc/security/limits.conf
[... 7 lines cut ...]
`Marco Accame 03 Nov 2014 11:55 AM`

```
With the above changes and after continuous use on iCubGenova04 the problem has not 
shown anymore.

The above changes related to robotIntenface (points 4 to 7) are already in icub-main.

The changes related to system (points 1 to 3) are issue of another task dealing with
preparation of a new live linux key.
Can anybody working on this task please add a link?
```
`NOTE1`: the iCubGenova04 of end 2014 has now become iCubNancy01.
`NOTE2`: The system settings of points 1 to 3 are now in use in our robots.

[... cut ...]
See https://redmine.robotology.eu/issues/561. 
We did many tests with the RT mode turned ON. It was decided:
1. the use periodic threads for both EthSender and EthReceiver (earlier the receiver was not periodic and was triggered on reception of any UDP packet) and,
2. to reduce the TX rate of the ETH boards to a reasonable value which is now configurable in runtime via xml files

We tested on old iCubGenova04, upper part of a new iCubGenova02, iCubHeidelberg, and later on we deployed on all other ETH robots.

lornat75, 2017-05-19:
I agree we should enable RT for the eth manager, also making sure it is always on by default and in future installations of the robot.

diegoferigo, 2017-05-25:
Thank you @marcoaccame for this detailed report, well done. Following especially your `Note-1` and `Note-2`, this is the priorities logic I propose:

```
|   50   | 51 ------- 59 | 60 ------------------- 79 | 80 ------------------ 99 |
  Kernel      Critic          Threads containing          Threads that should
  IRQs        general         applicatin code with        run on top of user
              threads         strict RT constraints       threads but w/o
                                                          strict RT constraints
```

Please let me know what your opinion about.

cc @marcoaccame @drdanz @lornat75 @traversaro @francesco-romano @pattacini @barbalberto

marcoaccame, 2017-05-25:
Hi @diegoferigo, it seems fine by me.

apaikan, 2017-05-25:
@diegoferigo  Please notice that the Ethernet controller thread in RT Linux has priority level of 50. 
When talking about FIFO scheduling, higher number has actually higher priority value. So increasing the application thread's priority higher than 50 can preempt Network thread and may result higher latency in general!  This is the main reason why the `yarprobotinterface` threads has priority (30) less than 50!

marcoaccame, 2017-05-25:
Hi @apaikan and @diegoferigo,

two things in parallel:

1. to be sure of the rule, hence to understand if higher number is also higher priority or if it is the other way round.
2. agree on the priority ladder.

About 1: I simply don't know.

About 2:  see also issue #1233.  My opinion is that from highest to lowest priorities: 
- Network thread: I agree, it must not be preempted 
- ethSender:  I cite from my NOTE-2 `(it normally executes for shorter time, plus we really want to deliver commands to the motors ASAP)`.
- ethReceiver: I cite from my NOTE-2: `(because it parses UDP packets, unlocks blocking requests and fills all data coming from the ETH boards inside the devices - embObjStrain, embObjMotionControl, etc. -, data which is normally read at 10 ms rate). `.
- wholeBodyDynamicsDevice: we want it to be very precise in time.
- yarprobotinterface: I cite from my NOTE-2: `Finally, as third I would put all other threads in yarprobotinterface (because I assume they just deal with port communication ... but see my Question-1).`

diegoferigo, 2017-05-25:
@apaikan 
I'm seeing here a big misconception, apparently there is a [big confusion](https://stackoverflow.com/questions/8887531/which-real-time-priority-is-the-highest-priority-in-linux) about how these ranges work, and it looks like you're right. The two scales use an opposite logic, and this invalidates the scale of the first post of this issue. I'll edit the different posts accordingly.

Thanks for your intervention, it was really helpful!

lornat75, 2017-05-26:
I suggest we start by rising priorities only of `ethReceiver` and `ethSender`. Changing priorities for other threads may have hidden side-effects, to it needs to be one with care.

diegoferigo, 2017-05-26:
I updated the first post, I hope now the situation is more clear to everyone.

@lornat75 
After the @apaikan comments, it seems that the current situation of priorities already looks good and it is compliant of the [ranges I envisioned](https://github.com/robotology/yarp/issues/1215#issuecomment-304004656).

During the last few days, the flag `ICUB_USE_REALTIME_LINUX` has been set to `ON` in the `iCubGenova04` and all worked flawlessly.

closed 2025-01-30 by diegoferigo