robotology/yarp #1218: URLs of Doxygen Pages are broken

traversaro, 2017-05-18:
Currently all the URL pointing to doxygen documentation are broken. 

For example, the link http://www.yarp.it/tutorials.html in https://github.com/robotology/yarp#tutorials returns 404 error. 
http://yarp.it seems to redirect to http://yarp.it/master , so it would be nice to redirect http://www.yarp.it/tutorials.html to http://www.yarp.it/master/tutorials.html, and so on so forth. 

cc @drdanz @mbrunettini

mbrunettini, 2017-05-18:
I am working on the documentation.
it's just a work in progress

traversaro, 2017-05-18:
Ok, let's keep this issue open for anyone that is experiencing the same problem.

traversaro, 2017-05-18:
Ok, I did not notice that apparently it is working now, closing.

closed 2017-05-18 by traversaro