robotology/yarp #1228: Link to yarp installation instructions is broken

gabrielenava, 2017-05-23:
Following the instructions in [this page](http://wiki.icub.org/wiki/Linux:Installation_from_sources), I was trying to open the link with the instructions that explain how to install yarp:

 http://wiki.icub.org/yarp/specs/dox/user/html/install_yarp_linux.html

but apparently the link is broken. Actually, it seems also this page does not exist anymore:

www.yarp.it/install.html

but since few weeks ago, I was able to reach it from my pc.

traversaro, 2017-05-23:
Duplicate of https://github.com/robotology/yarp/issues/1218 , @mbrunettini was working on that , what is the status? 
In the meanwhile you can access it at www.yarp.it/master/install.html

mbrunettini, 2017-05-23:
Link updated

closed 2017-05-23 by mbrunettini