robotology/yarp #1229: Errors in how yarp::dev::BoschIMU::run() handles race conditions

diegoferigo, 2017-05-23:
Referring to https://github.com/robotology/icub-support/issues/449 and https://github.com/robotology/icub-support/issues/454, it turns out that most of the time taken by `yarp::dev::WholeBodyDynamicsDevice::run()` is the `readSensors()` method, and particularly the IMU section. Sharing this with the team, we figured out that the problems related to the hardware fault started exactly when the IMUs switched to the new Bosch's ones.

In particular, the problems arise due to a race condition on the IMU data between the `BoshIMU::run()` and `BoshIMU::read()` methods.

So, by the end of the story, the thread priorities were not such a problem. However, this allowed us to figure out many forgotten things as reported in https://github.com/robotology/yarp/issues/1215.

cc @barbalberto 

FYI cc @DanielePucci @marcoaccame @serena-ivaldi

DanielePucci, 2017-05-23:
Very well done @diegoferigo Thanks a lot for the additional effort on this issue.

barbalberto, 2017-05-24:
I see... both `read` and `run` access to the same mutex to protect the data, but the `run` may take some time since it is reading the data from the serial line, so the `read` may be queued and take time as well.
This delay is affecting network wrapper functionality, but may be an issue when calling the function directly from a timed loop.

Do you thing adding an extra buffer to reduce the protected region to the minimum will improve this situation?

traversaro, 2017-05-24:
Behind the mutex problem, the fact that sometimes the Bosch IMU takes 100 ms to read its data is scaring by itself. It is possible that this is caused by communication issues?

Some issues related to the Bosch BNO055: 
  * https://github.com/robotology/icub-support/issues/402
  * https://github.com/robotology/icub-support/issues/358
  * https://github.com/robotology/yarp/issues/1229

diegoferigo, 2017-05-24:
@barbalberto 
I just opened PR #1231. Yes, I think it makes more sense if the methods that gather data from the sensors are not included inside the protected portion of code since they might be slow.

traversaro, 2017-05-30:
Fixed by https://github.com/robotology/yarp/issues/1231 .

closed 2017-05-30 by traversaro