robotology/yarp #1232: transmission rate in yarp from PC104 atom

reafrancesco, 2017-05-24:
Together with @barbalberto I have noticed an high variability of the transmission rate of packets from a pc104 atom (talking head) and the rest of standard iCub network.
We focused on "yarpdev --device test_grabber --width 320 --height 240" as reference test.
Whereas the test_grabber prints out the following messages
Read [152] frames in 5[s], average period 33.11[ms], min 33.25[ms], max 33.44[ms]
any yarpview (running on any other machine in the iCub network), once connected to the /grabber (in both tcp and udp) reports framerate that spans from ~3fps to ~30fps.

The pc104 atom is demanding less than 10% in this condition (measured with top).

Yarpview (running on pc104atom), once connected to the /grabber reports framerate that spans from ~27fps to ~32fps.

@lornat75 Could it be that a possible bugs is related to the protocol used by yarp to transfer packets when compiled for a pc104 atom. To find this out I am going to stream out video through the network with VLC or gstreamer. I am also wondering if other research group using pc104atom have experience similar downgrade in throughput.

lornat75, 2017-05-26:
This is a strange behavior, are you sure the network link is 1Gb/s? There may be something wrong in the cabling or network driver that downgrades it to 100Mb/s.

barbalberto, 2017-05-30:
Yesterday we experienced a similar issue on the Jetson Tegra TX2 with standard USB rgb camera, while the depth camera was wrking with correct 30fps.

Therefore we used the test_grabber also shown 10fps using 2 streaming in 640x480 with mjpeg carrier and 2fps with udp.

test-grabber run on the r1-base, the streaming was ok.

EliseiFrederic, 2017-09-08:
We also have a talking head version with pc104 atom here at Grenoble. In our experience, when doing only video with the camera, we have a sustained 30 fps. But when we have full robotic experience, many UDP packets are loss somewhere, be it sound or video frame (frame rate still peaks to 30 fps sometimes, but regularly slow down to 10 fps). We are using Bayer carrier with 640x480 and the 2 cameras at the same time.
 
We have been checking our 1Gb/s, both cables and connection hardware, but without finding the problem yet. If you have it too... maybe it's a software issue :-(

drdanz, 2020-07-14:
Can we close this issue?

closed 2020-07-28 by no-response[bot]