robotology/yarp #1238: Qt GUIs - Missing elements in Debian Stretch

kt10aan, 2017-05-27:
I am compiling YARP with GUIs in Debian Stretch (I know that it is ``testing`` but its been frozen since November).

Everything compiles fine, but ``YARPView`` fails to load with:

``
Warning: QT_DEVICE_PIXEL_RATIO is deprecated. Instead use: 
    QT_AUTO_SCREEN_SCALE_FACTOR to enable platform plugin controlled per-screen factors. 
    QT_SCREEN_SCALE_FACTORS to set per-screen factors. 
    QT_SCALE_FACTOR to set the application global scale factor.
QQmlApplicationEngine failed to load component
qrc:/qml/QtYARPView/main.qml:52 Type VideoSurface unavailable
qrc:/YARPView/VideoSurface.qml:304 Type FileDialog unavailable
qrc:/QtQuick/Dialogs/DefaultFileDialog.qml:47 module "Qt.labs.folderlistmodel" is not installed
qrc:/QtQuick/Dialogs/DefaultFileDialog.qml:48 module "Qt.labs.settings" is not installed
qrc:/QtQuick/Dialogs/DefaultFileDialog.qml:47 module "Qt.labs.folderlistmodel" is not installed
qrc:/QtQuick/Dialogs/DefaultFileDialog.qml:48 module "Qt.labs.settings" is not installed
``

The `` Qt`` version is ``5.7``, the default in Debian Stretch.
The  ``LD_LIBRARY_PATH`` is set to  ``$YARP_DIR/lib``

Is it an issue with the  ``Qt`` version or is there any missing package I need to install (I installed the ``Qt5`` packages that are mentioned in the [this page](http://www.yarp.it/install_yarp_linux.html)?

kt10aan, 2017-05-27:
Ok, I found that in Debian Stretch a few additional packages are required:

- qml-module-qt-labs-folderlistmodel
- qml-module-qt-labs-settings

With the packages above, ``YARPView`` works.

drdanz, 2017-05-27:
Great, they changed the names again! :man_facepalming: 
Thanks for reporting, I will update the instructions asap.
Reopening this until the documentation is updated...

commit 55f5125b (2017-06-06): Docs: Update documentation for debian stretch

Fixes: #1238
closed 2017-06-06 by drdanz