robotology/yarp #1246: [ControlBoardWrapper] Malformed ROS messages for ROS_only

francesco-romano, 2017-05-31:
If the `ControlBoardWrapper` uses ROS with the option `only`, then the resulting ROS topic messages are malformed. In particular the timestamp is not updated, and thus the messages are discarded by `/tf`

The time in the message is set at [this line](https://github.com/robotology/yarp/blob/master/src/libYARP_dev/src/devices/ControlBoardWrapper/ControlBoardWrapper.cpp#L930), but the time is updated only [here](https://github.com/robotology/yarp/blob/master/src/libYARP_dev/src/devices/ControlBoardWrapper/ControlBoardWrapper.cpp#L869), i.e. only if YARP is enabled.

randaz81, 2017-05-31:
I remember this was already discussed and fixed in the past months. Is it a regression bug? @barbalberto

francesco-romano, 2017-05-31:
I am afraid that we discussed about it, but nobody did a fix for it.

closed 2017-06-05 by drdanz