robotology/yarp #1252: yarplogger freezes after startup

S-Dafarra, 2017-06-05:
After launching the logger from the ``yarpmanager`` or the terminal, the logger freezes when pushing some buttons (for example the refresh one) and it needs to be killed.

@Nicogene @gabrielenava @francesco-romano

Nicogene, 2017-06-05:
Hi @S-Dafarra ,
I tried from terminal and `yarpmanager` both and it seems to work on my pc.
Are you on windows or linux?

francesco-romano, 2017-06-05:
We tried on two different linux machines and in both cases `yarplogger` freezes.

Nicogene, 2017-06-05:
At first glance `YarpLogger` doesn't seem to set any timeout for the connections, probably in the setup there are some dead ports and yarp logger when you press `refresh` awaits forever a reply from those ports

S-Dafarra, 2017-06-06:
It seems working smoothly now! May some of the latest commits have affected this issue?

drdanz, 2017-06-06:
Perhaps the same race condition affecting yarpmanager?

Nicogene, 2017-06-06:
For now I close it, if this happens again, tell us and we will investigate!

closed 2017-06-06 by Nicogene