robotology/yarp #1253: yarpmanager crashes continuosly

S-Dafarra, 2017-06-05:
The logger crashed with ``SEGFAULT`` a number of times, especially when pressing ``STOP ALL``, ``KILL ALL`` and ``RUN ALL``. This happens quite often, even if not all the times. I am on a Linux machine.

Nicogene, 2017-06-05:
The logger crashes or `yarpmanager`?
Speaking about `yarpmanager` last Thursday we fixed a bug that potentially can lead to segfault.
Try to pull. compile and install and let me know if you still have these problems

S-Dafarra, 2017-06-05:
Unfortunately ``yarpmanager``. I pulled this morning and I already tried to recompile but nothing changed.

Nicogene, 2017-06-05:
It's not `yarpmanager` fault, with 636554500a05a4a9173eb8a1a57a319a3743979b, we introduced a call to `yarp::os::Carriers::listCarriers` every time a port is registered and unregistered, but `Carriers` doesn't seem to be *multi-thread proof*, `RUN ALL` and `STOP ALL` stress this race condition that lead to `segfault`

closed 2017-06-05 by drdanz

drdanz, 2017-06-05:
@S-Dafarra Should be fixed now, please let me know if you still experience this issue.

commit 44a555c9 (2017-06-05): OS/YarpPluginSelector: Protect variable access using a mutex

Fixes: #1253