robotology/yarp #1265: Module does not start when defining YARP_CLOCK

S-Dafarra, 2017-06-13:
I am trying to test a module I have written using gazebo, but if I append ``YARP_CLOCK=/clock`` before launching it, the module get stuck after the following message
```
yarp: Anonymous port active at tcp://10.255.111.71:38400/
```
If I launch it without the clock port it starts smoothly.

barbalberto, 2017-06-13:
If you `yarp read` the `/clock` port, is the time being published correctly?

traversaro, 2017-06-13:
Are you in `devel` branch of yarp?

S-Dafarra, 2017-06-13:
@traversaro yes! I'm​ in devel branch.

francesco-romano, 2017-06-13:
I managed to replicate the problem with this simple code:

```cpp
#include <yarp/os/Network.h>
#include <yarp/os/Port.h>
#include <yarp/os/Time.h>

#include <iostream>
int main() {

    yarp::os::Network network;
    if (!yarp::os::Network::checkNetwork(5)) {
        std::cerr << "No network\n";
        return -1;
    }

    std::cerr << "Creating port\n";
    yarp::os::Port port;
    std::cerr << "Opening port\n";
    port.open("/test");
    std::cerr << "Port opened\n";

    for (int i = 0; i < 5; ++i) {
        std::cerr << ".";
        yarp::os::Time::delay(0.5);
    }
    std::cerr << std::endl;

    port.close();

    return 0;
}

```

Tomorrow I will git bisect to find the problem. It is a bit cumbersome as I have to update both yarp and gazebo plugins

francesco-romano, 2017-06-13:
Working commit: 397f21215a33cbfa4f80cd965aabc2eab1ab9c9a
Not working commit: 9256d7609933b7c08871b74bbacc1763f9ef9357

francesco-romano, 2017-06-13:
Result of bisect:

```
636554500a05a4a9173eb8a1a57a319a3743979b is the first bad commit
commit 636554500a05a4a9173eb8a1a57a319a3743979b
Author: Nicolo' Genesio <nicolo.genesio@iit.it>
Date:   Thu Jun 1 16:26:38 2017 +0200

    NameClient: fix yarp name /port accepts

:040000 040000 0dc8ca66078690338f8f27c2554b53845e97b494 9747e9811af36418072792c66c0cc877a4a50261 M	src
```

francesco-romano, 2017-06-13:
Tested also by checking out the last devel commit and manually rolled back 636554500a05a4a9173eb8a1a57a319a3743979b 
Without that commit the clock works

closed 2017-06-14 by drdanz

Nicogene, 2017-06-14:
Fixed in #1268

PR #1268 (2017-06-14): libYARP_OS-YarpPlugin: change Time::now() with SystemClock::nowSystem()
This PR fixes #1265.

Please review code.