robotology/yarp #1282: yarpmotorgui "get current range" error

gsaponaro, 2017-06-27:
Hi all, we have all the repositories updated to the **master** branch.

When launching **yarpmotorgui** it prints the following errors, for each robot part and for each joint:
```
[INFO]head iencs->getEncoders() ok!
[ERROR]Error while getting current range, part  head  joint  0 
[ERROR]Error while getting current range, part  head  joint  1 
[ERROR]Error while getting current range, part  head  joint  2 
[ERROR]Error while getting current range, part  head  joint  3 
[ERROR]Error while getting current range, part  head  joint  4 
[ERROR]Error while getting current range, part  head  joint  5 
```
[full log here](https://gist.github.com/gsaponaro/5d234d63fd5c671474a5c2169c9c7fe4)

When the above happens, **yarprobotinterface** prints the following:
```
yarp: Receiving input from /yarpmotorgui0//icub/head/rpc:o to /icub/head/rpc:i using tcp
yarp: Receiving input from /yarpmotorgui0//icub/head/command:o to /icub/head/command:i using udp
yarp: Sending output from /icub/head/stateExt:o to /yarpmotorgui0//icub/head/stateExt:i using udp
Registering new thread 8 out of 500
[ERROR]getCurrentRangeRaw not yet implemented for CanBusMotionControl
[ERROR]getCurrentRangeRaw not yet implemented for CanBusMotionControl
[ERROR]getCurrentRangeRaw not yet implemented for CanBusMotionControl
[ERROR]getCurrentRangeRaw not yet implemented for CanBusMotionControl
[ERROR]getCurrentRangeRaw not yet implemented for CanBusMotionControl
[ERROR]getCurrentRangeRaw not yet implemented for CanBusMotionControl
```

However, after seeing these errors, we are able to control the robot normally with yarpmotorgui.

/cc @vicentepedro

francesco-romano, 2017-06-27:
Hi @gsaponaro 
Those errors are "normal". They do not cause any problem to the motorgui.
I think they are there for future features.

gsaponaro, 2017-06-28:
I see, thanks. Feel free to close the issue, if you wish.

closed 2017-06-28 by drdanz