robotology/yarp #1285: "using namespace std" directive in several headers

aerydna, 2017-06-28:
i found the directive `using namespace std;` in the headers of libYARP_manager and yarpmanager (plus one in the mainwindow.h of yarpdataplayer).

this can cause several problem.. for instance by including <vector> or <map> in ResurceFinder.h arise some ambiguities between std and libYARP_OS.

francesco-romano, 2017-06-28:
I totally agree.
`using  namespace` (in general, not only `std`) should be prohibited at least in library headers.

lornat75, 2017-06-28:
should --> must be prohibited! Please fix all library header files which contain the `using` directive.

Nicogene, 2017-07-12:
Fixed by #1294

closed 2017-07-13 by drdanz

pattacini, 2017-07-13:
it's not compiling... just fixing

pattacini, 2017-07-13:
the problem was present only in `devel` after merging.
Fixed.

drdanz, 2017-07-13:
@pattacini Thanks!

PR #1294 (2017-07-13): Remove using namespace directive from headers 
This PR remove `using namespace` directive from several headers.
Fixes #1285
Please review code.