robotology/yarp #1292: roslaunch do not run if launched with the `localhost` deployer

traversaro, 2017-07-11:
Apparently, it is not possible to "launch" the `roslaunch` ROS command through the localhost deployer of the yarpmanager. Launching it through a yarprun server works fine, but if you try to launch it from with the localhost deployer you get the following error: 
~~~
[ERR] (appName) cannot run roslaunch on localhost : 
~~~
OS : Ubuntu 16.04 
ROS Distribution: Kinetic 
YARP version : https://github.com/robotology/yarp/commit/a41dacadc54828b48eb5f781cd279fd65ebddd1b

traversaro, 2017-07-12:
I suspect that the problem is that `roslaunch` is a python program.

traversaro, 2017-07-12:
Indeed, I have verified that the problem is the inability to launch python scripts.

traversaro, 2017-07-12:
Interesting, both `yarprun` [1] and yarpmanager's `LocalBroker` [2] execute the "child" program using the standard unix call `execvp`, so it is strange that one works while the other is not working. 

Note that a possible workaround is to run roslaunch as ``python `which roslaunch` ``, but the problem would be that  `` `which roslaunch` `` cannot be run by the yarpmanager/yarprun. 

[1] : https://github.com/robotology/yarp/blob/9a395f6e41e01d8d350b6a4e5e2b2f6c6f913b74/src/libYARP_OS/src/Run.cpp#L2740 
[2] : https://github.com/robotology/yarp/blob/9a395f6e41e01d8d350b6a4e5e2b2f6c6f913b74/src/libYARP_manager/src/localbroker.cpp#L942

Nicogene, 2017-08-28:
Hi @traversaro, have you tried with `SKIP_ACE`?
In `src/libYARP_OS/include/yarp/os/impl/PlatformUnistd.h` we have:
```
namespace yarp {
namespace os {
namespace impl {

#if defined(YARP_HAS_ACE)
    using ACE_OS::rmdir;
    inline int gethostname(char *name, size_t len) { return ACE_OS::hostname(name, len); }
    using ACE_OS::getpid;
    using ACE_OS::getppid;
    using ACE_OS::isatty;
    using ACE_OS::getcwd;
    using ACE_OS::fork;
    using ACE_OS::pipe;
    using ACE_OS::dup;
    using ACE_OS::dup2;
    using ACE_OS::execvp;
    using ACE_OS::chdir;
    using ACE_OS::unlink;

    // ACE version of execl/execlp/execle are just fake implementation, see
    // https://github.com/DOCGroup/ACE_TAO/issues/409
    // https://github.com/DOCGroup/ACE_TAO/blob/ACE%2BTAO-6_4_3/ACE/ace/OS_NS_unistd.cpp#L227
    // (last ACE version tested: 6.4.3).
    // This is not a big issue since (at the moment) it is used only in
    // #if !defined(_WIN32) branches, but we might need to fix this at some
    // point.
    using ::execlp;
#else
    using ::rmdir;
    using ::gethostname;
    using ::getpid;
    using ::getppid;
    using ::isatty;
    using ::getcwd;
    using ::fork;
    using ::pipe;
    using ::dup;
    using ::dup2;
    using ::execlp;
    using ::execvp;
    using ::chdir;
    using ::unlink;
#endif

} // namespace impl
} // namespace os
} // namespace yarp
```
 `yarp::os::impl::execvp` is the same called in localbroker if yarp is compiled without ACE, otherwise in Run.cpp is `ACE_OS::execvp`.
Do you think that the difference of behaviour could be caused by this?

traversaro, 2017-08-28:
In the system I was testing, `SKIP_ACE` was set to `OFF` . Were you able to reproduce the problem?

drdanz, 2017-08-29:
After a little debugging I noticed that

1) Fails
```xml
    <module>
        <name>/usr/bin/python</name>
        <parameters>-c "print 2"</parameters>
        <node>localhost</node>
        <prefix></prefix>
    </module>
```

2) Works
```xml
    <module>
        <name>/usr/bin/python</name>
        <parameters>-c "import time; time.sleep(5)"</parameters>
        <node>localhost</node>
        <prefix></prefix>
    </module>
```

The same is true for bash, works if I start sleep but fails for other commands. Other commands that exit instantaneously (env, ls, pwd) all fail.

So I created this script

```py
#!/usr/bin/env python

import time
time.sleep(5)
```

and started it with

```xml
    <module>
        <name>/home/drdanz/test.py</name>
        <node>localhost</node>
        <prefix></prefix>
    </module>
```

and this works as well.

**Conclusion**: I think that the problem is not that the script is not started, but that for some reason it exits immediately. There is probably some second check after starting the app that ensures if the app is started that is executed too late (the app is already exited).

I'm not sure if this is the same issue as `roslaunch`, but perhaps it forks immediately and exits?

Nicogene, 2018-03-06:
I tried to run a python script also with `SKIP_ACE` to ON and the result is the same, so it is not ACE's fault (as expected).

traversaro, 2018-03-06:
cc @GiuliaP

Yeshasvitvs, 2019-01-25:
@traversaro @Nicogene  Is there any update on this  issue ? 
Is there a way to launch `roslaunch` through `yarpmanager` application ??

Nicogene, 2019-02-04:
> I'm not sure if this is the same issue as roslaunch, but perhaps it forks immediately and exits?

If it is confirmed that `roslaunch` forks immediately, then the problem should be only on checking if the process is running or not, but the script should be launched anyway. Am I right?