robotology/yarp #1309: Setting the PWM through the ControlBoardRemapper is not working

nunoguedelha, 2017-08-03:
### Context
A control board re-mapper (class `ControlBoardRemapper)` remaps the axes (joint motors) from a set of control boards. After attaching the respective instanciated device drivers, the re-mapper can access the control methods implemented by these devices. The methods for setting directly the motor **PWM** are grouped in the interface class `IPWMControl`.

### The issue
While trying to set the PWM using the re-mapper, we noticed that the pointer to the device driver for accessing the PWM control methods, returned by the control board re-mapper, was `NULL`, causing a segmentation fault.
This occured when implementing the YARP bindings for the `IPWMControl` methods.

### Root cause
The `IPWMControl` interface was not implemented in the `ControlBoardRemapper` class.

nunoguedelha, 2017-08-03:
This is fixed by PR https://github.com/robotology/yarp/pull/1310, currently under review.

traversaro, 2017-08-07:
Fixed by https://github.com/robotology/yarp/pull/1310 .

closed 2017-08-07 by traversaro