robotology/yarp #1318: /ros port removed by yarprun

randaz81, 2017-08-11:
Similar to #1134
Try the following:

- roscore
- yarp server --ros (the nameserver will register the /ros port) 
- yarprun --server /ros   **(ARGHHH THAT'S BAD! No error message displayed!)**
- killall yarprun (gracefully quitting, no -9 signal is used, port /ros will be closed by yarprun)

now /ros port is no more registered on the nameserver...

drdanz, 2017-08-11:
Wow this is cool, it works also for the name server
* yarp server (assuming default namespace /root)
* yarprun --server /root
* killall yarprun
yarpserver /root registration disappears

Nicogene, 2017-09-05:
I reproduced this issue also with: `yarp write /root`, `yarp read /root`, `yarpview --name /root`, and probably it happens also with other executables...  I'm afraid that is more toxic than it seems :fearful:.
`NetworkBase::exists()` return true, but I think that is not checked in `port.open()` and it return successfully

Tobias-Fischer, 2018-05-10:
So it seems like this bug is not really related to ROS then, is it? The `/ros` port is just a special case. Probably a good idea to rename the title.

Nicogene, 2018-05-10:
It is somehow ROS related because `/ros` is not a "yarp" port, so it respond differently when pinged.
Consequences:
- `yarp clean` remove it because it doesn't respond as expected.
- Every other executable that try to open /ros will overwrite it in the nameserver because it doesn't trigger the address conflict.

I think that the portname validation is the only way to fix all these problems :thinking: