robotology/yarp #1323: YARP release v2.3.70 does not compile if  ENABLE_yarpcar_xmlrpc is ON

liesrock, 2017-08-22:
I tried to compile the latest YARP release at https://github.com/robotology/yarp/tree/v2.3.70 , but I had to put the CMake variable ENABLE_yarpcar_xmlrpc OFF otherwise I was receiving the following compilation error:

```
[ 59%] Building CXX object src/carriers/xmlrpc_carrier/CMakeFiles/yarp_xmlrpc.dir/yarp_plugin_yarpcar_xmlrpc.cpp.o
In file included from /home/lucamuratore/advr-superbuild/external/YARP/src/carriers/xmlrpc_carrier/XmlRpcStream.h:17:0,
                 from /home/lucamuratore/advr-superbuild/external/YARP/src/carriers/xmlrpc_carrier/XmlRpcCarrier.h:12,
                 from /home/lucamuratore/advr-superbuild/build/external/YARP/src/carriers/xmlrpc_carrier/yarp_plugin_yarpcar_xmlrpc.cpp:10:
/opt/ros/indigo/include/XmlRpcClient.h:70:18: warning: ‘virtual void XmlRpc::XmlRpcClient::close()’ can be marked override [-Wsuggest-override]
     virtual void close();
                  ^
/opt/ros/indigo/include/XmlRpcClient.h:75:22: warning: ‘virtual unsigned int XmlRpc::XmlRpcClient::handleEvent(unsigned int)’ can be marked override [-Wsuggest-override]
     virtual unsigned handleEvent(unsigned eventType);
                      ^
In file included from /home/lucamuratore/advr-superbuild/external/YARP/src/carriers/xmlrpc_carrier/XmlRpcStream.h:18:0,
                 from /home/lucamuratore/advr-superbuild/external/YARP/src/carriers/xmlrpc_carrier/XmlRpcCarrier.h:12,
                 from /home/lucamuratore/advr-superbuild/build/external/YARP/src/carriers/xmlrpc_carrier/yarp_plugin_yarpcar_xmlrpc.cpp:10:
/opt/ros/indigo/include/XmlRpcServerConnection.h:50:22: warning: ‘virtual unsigned int XmlRpc::XmlRpcServerConnection::handleEvent(unsigned int)’ can be marked override [-Wsuggest-override]
     virtual unsigned handleEvent(unsigned eventType);
                      ^
In file included from /home/lucamuratore/advr-superbuild/external/YARP/src/carriers/xmlrpc_carrier/XmlRpcCarrier.h:12:0,
                 from /home/lucamuratore/advr-superbuild/build/external/YARP/src/carriers/xmlrpc_carrier/yarp_plugin_yarpcar_xmlrpc.cpp:10:
/home/lucamuratore/advr-superbuild/external/YARP/src/carriers/xmlrpc_carrier/XmlRpcStream.h:32:5: error: ‘YarpXmlRpc’ does not name a type
     YarpXmlRpc::XmlRpcClient client;
     ^
/home/lucamuratore/advr-superbuild/external/YARP/src/carriers/xmlrpc_carrier/XmlRpcStream.h:33:5: error: ‘YarpXmlRpc’ does not name a type
     YarpXmlRpc::XmlRpcServerConnection server;
     ^
/home/lucamuratore/advr-superbuild/external/YARP/src/carriers/xmlrpc_carrier/XmlRpcStream.h: In constructor ‘yarp::os::XmlRpcStream::XmlRpcStream(yarp::os::TwoWayStream*, bool, bool)’:
/home/lucamuratore/advr-superbuild/external/YARP/src/carriers/xmlrpc_carrier/XmlRpcStream.h:41:13: error: class ‘yarp::os::XmlRpcStream’ does not have any field named ‘client’
             client("notset", 0),
             ^
/home/lucamuratore/advr-superbuild/external/YARP/src/carriers/xmlrpc_carrier/XmlRpcStream.h:42:13: error: class ‘yarp::os::XmlRpcStream’ does not have any field named ‘server’
             server(0, YARP_NULLPTR),
             ^
/home/lucamuratore/advr-superbuild/external/YARP/src/carriers/xmlrpc_carrier/XmlRpcStream.h:47:9: error: ‘client’ was not declared in this scope
         client.reset();
         ^
/home/lucamuratore/advr-superbuild/external/YARP/src/carriers/xmlrpc_carrier/XmlRpcStream.h:48:9: error: ‘server’ was not declared in this scope
         server.reset();
         ^
```

@DanielePucci @gabrielenava @diegoferigo

liesrock, 2017-08-22:
I checked that TRAVIS build passes https://travis-ci.org/robotology/yarp/jobs/243258968 ... tell me if I am missing something.

traversaro, 2017-08-22:
On which OS are you compiling YARP? 
I suspect there is some kind of incompatibility between the version of `xmlrpcpp` installed by ROS Indigo and the one used by YARP. 
You can try to force using the YARP-internal version of `xmlrpcpp` by setting the YARP_USE_SYSTEM_xmlrpcpp CMake variable to `FALSE`.

traversaro, 2017-08-22:
I actually checked the code and apparently `xmlrpccpp` is always compiled: https://github.com/robotology/yarp/blob/v2.3.70/cmake/YarpFindDependencies.cmake#L353 . In this case, I think there is something wrong in the order of the include paths. Can you run `make VERBOSE=1` to get the actual command with which the compiler was invoked?

liesrock, 2017-08-22:
Thanks @traversaro , sorry i did not specify it: it is Ubuntu 14.04 
I tried to force using the YARP-internal version of `xmlrpcpp` by setting the `YARP_USE_SYSTEM_xmlrpcpp` CMake variable to `FALSE`., but it gives the same error.

liesrock, 2017-08-22:
Here it is the output of the verbose compilation on the target:

```
Building CXX object src/carriers/xmlrpc_carrier/CMakeFiles/yarp_xmlrpc.dir/yarp_plugin_yarpcar_xmlrpc.cpp.o
cd /home/lucamuratore/advr-superbuild/build/external/YARP/src/carriers/xmlrpc_carrier && /usr/bin/c++   -DYARP_PRESENT -D_REENTRANT -Dyarp_xmlrpc_EXPORTS -I/home/lucamuratore/advr-superbuild/build/external/YARP/src/carriers/xmlrpc_carrier -I/home/lucamuratore/advr-superbuild/external/YARP/src/carriers/xmlrpc_carrier -isystem /home/lucamuratore/advr-superbuild/external/YARP/extern/xmlrpcpp/xmlrpcpp/src -I/home/lucamuratore/advr-superbuild/external/YARP/src/libYARP_OS/include -I/home/lucamuratore/advr-superbuild/build/external/YARP/src/libYARP_conf/include  -Wall -Wextra -Wsign-compare -Wpointer-arith -Winit-self -Wnon-virtual-dtor -Wcast-align -Wunused -Wunused-but-set-variable -Wvla -Wmissing-include-dirs -Wlogical-op -Wreorder -Wsizeof-pointer-memaccess -Woverloaded-virtual -Wundef -Wredundant-decls -Wunknown-pragmas -Wunused-result -Wc++11-compat -Wsuggest-override  -Wno-unused-parameter -Wno-long-long -Wno-cast-align  -fvisibility=hidden -fvisibility-inlines-hidden  -Wdeprecated-declarations -O3 -DNDEBUG -fPIC   -std=c++11 -o CMakeFiles/yarp_xmlrpc.dir/yarp_plugin_yarpcar_xmlrpc.cpp.o -c /home/lucamuratore/advr-superbuild/build/external/YARP/src/carriers/xmlrpc_carrier/yarp_plugin_yarpcar_xmlrpc.cpp
```

traversaro, 2017-08-22:
That is strange. 
The compiler complains about a include file included from the `/opt/ros/indigo` directory: 
~~~
In file included from /home/lucamuratore/advr-superbuild/external/YARP/src/carriers/xmlrpc_carrier/XmlRpcStream.h:18:0,
                 from /home/lucamuratore/advr-superbuild/external/YARP/src/carriers/xmlrpc_carrier/XmlRpcCarrier.h:12,
                 from /home/lucamuratore/advr-superbuild/build/external/YARP/src/carriers/xmlrpc_carrier/yarp_plugin_yarpcar_xmlrpc.cpp:10:
/opt/ros/indigo/include/XmlRpcServerConnection.h:50:22: warning: ‘virtual unsigned int XmlRpc::XmlRpcServerConnection::handleEvent(unsigned int)’ can be marked override [-Wsuggest-override]
     virtual unsigned handleEvent(unsigned eventType);
                      ^
~~~
but that direction is not in the include directories passed to the compiler through the command line. I wonder if there is some strange environmental setting that overwrites the path passed through command line. 
Can you please paste:
* The content of `/home/lucamuratore/advr-superbuild/build/external/YARP/src/carriers/xmlrpc_carrier/yarp_plugin_yarpcar_xmlrpc.cpp` (to make sure there is not absolute path in the include
* The output of `cpp -v /dev/null -o /dev/null` (see https://gcc.gnu.org/onlinedocs/cpp/Search-Path.html)

barbalberto, 2017-08-22:
Probably the ros `setup.bash` expands the environment. Just for testing, can you avoid to source of ros setup, delete the yarp build folder and compile it from clean? This way we can detect if there is some interferences

gabrielenava, 2017-08-22:
I disabled the following line in the .bashrc:

```
. /opt/ros/indigo/setup.bash
```
and now yarp compiles even with `ENABLE_yarpcar_xmlrpc` setted to ON, so I think that environmental setting was causing the issue

traversaro, 2017-08-22:
@gabrielenava Please run `cpp -v /dev/null -o /dev/null` so we can understand the root of this issue and avoid relying on workaround, thanks.

traversaro, 2017-08-22:
Related issue: https://github.com/ros/catkin/pull/758 . Apparently it is solved in Kinetic.

traversaro, 2017-08-22:
A more permanent workaround is to remove the `SYSTEM` keyword from https://github.com/robotology/yarp/blob/devel/src/carriers/xmlrpc_carrier/CMakeLists.txt#L15 , but I guess this would probably cause some warnings (see https://github.com/robotology/yarp/commit/a7b6ce9b32393621892bcc6b471f93b7386c726d).

francesco-romano, 2017-08-22:
@traversaro why do you think `SYSTEM` causes this clash in include directories?

traversaro, 2017-08-22:
The directories specified with `CPATH` are searched before any path specified with `-isystem`, see https://gcc.gnu.org/onlinedocs/cpp/Environment-Variables.html and https://gcc.gnu.org/onlinedocs/cpp/Invocation.html#Invocation .

gabrielenava, 2017-08-22:
I reverted my previous modification, updated the cache and run `cpp -v /dev/null -o /dev/null` in a terminal. This is the output:

```

lucamuratore@walkman-pilot-pc:~$ cpp -v /dev/null -o /dev/null
Using built-in specs.
COLLECT_GCC=cpp
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 4.8.4-2ubuntu1~14.04.3' --with-bugurl=file:///usr/share/doc/gcc-4.8/README.Bugs --enable-languages=c,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-4.8 --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --with-gxx-include-dir=/usr/include/c++/4.8 --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --enable-gnu-unique-object --disable-libmudflap --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-4.8-amd64/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-4.8-amd64 --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-4.8-amd64 --with-arch-directory=amd64 --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --enable-objc-gc --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --with-tune=generic --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 4.8.4 (Ubuntu 4.8.4-2ubuntu1~14.04.3) 
COLLECT_GCC_OPTIONS='-E' '-v' '-o' '/dev/null' '-mtune=generic' '-march=x86-64'
 /usr/lib/gcc/x86_64-linux-gnu/4.8/cc1 -E -quiet -v -imultiarch x86_64-linux-gnu /dev/null -o /dev/null -mtune=generic -march=x86-64 -fstack-protector -Wformat -Wformat-security
ignoring nonexistent directory "/usr/local/include/x86_64-linux-gnu"
ignoring nonexistent directory "/usr/lib/gcc/x86_64-linux-gnu/4.8/../../../../x86_64-linux-gnu/include"
#include "..." search starts here:
#include <...> search starts here:
 /opt/ros/indigo/include
 /usr/lib/gcc/x86_64-linux-gnu/4.8/include
 /usr/local/include
 /usr/lib/gcc/x86_64-linux-gnu/4.8/include-fixed
 /usr/include/x86_64-linux-gnu
 /usr/include
End of search list.
COMPILER_PATH=/usr/lib/gcc/x86_64-linux-gnu/4.8/:/usr/lib/gcc/x86_64-linux-gnu/4.8/:/usr/lib/gcc/x86_64-linux-gnu/:/usr/lib/gcc/x86_64-linux-gnu/4.8/:/usr/lib/gcc/x86_64-linux-gnu/
LIBRARY_PATH=/usr/lib/gcc/x86_64-linux-gnu/4.8/:/usr/lib/gcc/x86_64-linux-gnu/4.8/../../../x86_64-linux-gnu/:/usr/lib/gcc/x86_64-linux-gnu/4.8/../../../../lib/:/lib/x86_64-linux-gnu/:/lib/../lib/:/usr/lib/x86_64-linux-gnu/:/usr/lib/../lib/:/usr/lib/gcc/x86_64-linux-gnu/4.8/../../../:/lib/:/usr/lib/
COLLECT_GCC_OPTIONS='-E' '-v' '-o' '/dev/null' '-mtune=generic' '-march=x86-64'

```

drdanz, 2017-08-29:
As a workaround, I think you can do something like this:

```bash
. /opt/ros/indigo/setup.bash
unset CPATH
```

The bug was fixed upstream, so I think we can close this?

traversaro, 2017-08-29:
Unfortunately Indigo is still quite widespread, but I don't think there is anything more we can do (we could check in CMake if `CPATH` contains the indigo stuff and print a warning in that case, but it seems quite dangerous) , so I think it is ok to close this issue.

drdanz, 2017-08-29:
Closed, please re-open if you think there is more to say about this

closed 2017-08-29 by drdanz