robotology/yarp #1325: Yarp devel cannot be compiled with binding

towardthesea, 2017-08-25:
The updated of devel branch of yarp cannot be compiled when I turn the binding on. The following is part of log when compiling with `Python binding`
```
/usr/local/src/robot/yarp/src/libYARP_sig/include/yarp/sig/Image.h:654: Warning 512: Overloaded method yarp::sig::ImageOf< yarp::sig::PixelInt >::safePixel(int,int) const ignored,
/usr/local/src/robot/yarp/src/libYARP_sig/include/yarp/sig/Image.h:654: Warning 512: using non-const method yarp::sig::ImageOf< yarp::sig::PixelInt >::safePixel(int,int) instead.
/usr/local/src/robot/yarp/src/libYARP_sig/include/yarp/sig/Image.h:650: Warning 512: Overloaded method yarp::sig::ImageOf< yarp::sig::PixelFloat >::safePixel(int,int) const ignored,
/usr/local/src/robot/yarp/src/libYARP_sig/include/yarp/sig/Image.h:650: Warning 512: using non-const method yarp::sig::ImageOf< yarp::sig::PixelFloat >::safePixel(int,int) instead.
/usr/local/src/robot/yarp/src/libYARP_sig/include/yarp/sig/Image.h:651: Warning 512: Overloaded method yarp::sig::ImageOf< yarp::sig::PixelRgbFloat >::safePixel(int,int) const ignored,
/usr/local/src/robot/yarp/src/libYARP_sig/include/yarp/sig/Image.h:651: Warning 512: using non-const method yarp::sig::ImageOf< yarp::sig::PixelRgbFloat >::safePixel(int,int) instead.
bindings/CMakeFiles/_yarp_python.dir/build.make:61: recipe for target 'lib/python/yarpPYTHON_wrap.cxx' failed
make[2]: *** [lib/python/yarpPYTHON_wrap.cxx] Error 22
make[2]: *** Deleting file 'lib/python/yarpPYTHON_wrap.cxx'
CMakeFiles/Makefile2:4440: recipe for target 'bindings/CMakeFiles/_yarp_python.dir/all' failed
make[1]: *** [bindings/CMakeFiles/_yarp_python.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2
```
The full log of compilation is [yarp_build.txt](https://github.com/robotology/yarp/files/1251757/yarp_build.txt)

Cheers,
Phuong

towardthesea, 2017-08-25:
Thanks @traversaro . Waiting for the merge!

traversaro, 2017-08-25:
Fixed by https://github.com/robotology/yarp/pull/1326 .

closed 2017-08-25 by traversaro