robotology/yarp #1335: yarp devel is not configuring if YCM 0.4.0 is installed in the system

traversaro, 2017-09-04:
I have `ycm` installed from homebrew, and the latest YARP devel is not configuring with this error:
~~~
CMake Error at /usr/local/share/YCM/modules/AddInstallRPATHSupport.cmake:137 (file):
  file RELATIVE_PATH must be passed a full path to the directory:
  INSTALL_NAME_DIR
Call Stack (most recent call first):
  cmake/YarpOptions.cmake:359 (add_install_rpath_support)
  CMakeLists.txt:63 (include)
~~~

traversaro, 2017-09-04:
The problem seems related to https://github.com/robotology/yarp/pull/1322 . 
cc @francesco-romano @drdanz

traversaro, 2017-09-04:
A possible workaround is to simply uninstall the homebrew ycm : `brew uninstall ycm` .

traversaro, 2017-09-04:
I suspect bumping https://github.com/robotology/yarp/blob/devel/CMakeLists.txt#L16 to the YCM version that will be synced with YARP will be sufficient to solve the issue.

francesco-romano, 2017-09-04:
Ok. Let's wait for the bump. 
If needed we can update the YCM formula.

diegoferigo, 2017-09-06:
This bug affects also linux

drdanz, 2017-09-07:
A possible workaround is to bump YCM required version in YARP...

traversaro, 2017-09-09:
@drdanz What is the next version of YCM? I think we should bump to that version.

drdanz, 2017-09-21:
Fixed in 89573412ff2e787a14aca863417d57e066059782

commit 89573412 (2017-09-21): Update YCM to latest devel (0.5.1)

Fixes #1335
closed 2017-09-21 by drdanz