robotology/yarp #1341: YARP devel still not compiling with python bindings

Kuronogard, 2017-09-08:
Related to: https://github.com/robotology/yarp/issues/1325

When generating and building the project in "v2.3.70" and "devel" with python bindings:
```bash
cmake .. -DYARP_COMPILE_BINDINGS=ON -DCREATE_PYTHON=ON
make
```
I get the following outputs:
### v2.3.70
```bash
/home/dennis/repos/yarp/src/libYARP_OS/include/yarp/os/ConstString.h:192: Warning 389: operator[] ignored (consider using %extend)
/home/dennis/repos/yarp/src/libYARP_OS/include/yarp/os/ConstString.h:193: Warning 389: operator[] ignored (consider using %extend)
/home/dennis/repos/yarp/src/libYARP_OS/include/yarp/os/ConstString.h:273: Warning 321: 'hash' conflicts with a built-in name in python
/home/dennis/repos/yarp/src/libYARP_OS/include/yarp/os/Portable.h:31: Error: Syntax error in input(3).
make[2]: *** [lib/python/yarpPYTHON_wrap.cxx] Error 1
make[1]: *** [bindings/CMakeFiles/_yarp_python.dir/all] Error 2
make: *** [all] Error 2
```
Error located in Portable.h:31

### devel
```bash
/home/dennis/repos/yarp/src/libYARP_OS/include/yarp/os/ConstString.h:192: Warning 389: operator[] ignored (consider using %extend)
/home/dennis/repos/yarp/src/libYARP_OS/include/yarp/os/ConstString.h:193: Warning 389: operator[] ignored (consider using %extend)
/home/dennis/repos/yarp/src/libYARP_OS/include/yarp/os/ConstString.h:273: Warning 321: 'hash' conflicts with a built-in name in python
/home/dennis/repos/yarp/src/libYARP_OS/include/yarp/os/Contact.h:83: Error: Syntax error in input(3).
make[2]: *** [lib/python/yarpPYTHON_wrap.cxx] Error 1
make[1]: *** [bindings/CMakeFiles/_yarp_python.dir/all] Error 2
make: *** [all] Error 2
```
Error located in Contact.h:83

Kuronogard, 2017-09-09:
By the way, that is happening in an almost-fresh install of **Ubuntu 14.04**

PeterBowman, 2017-09-09:
I can't reproduce this on Ubuntu 16.04 at `v2.3.70` and current `master` (`devel` is failing due to #1335) with Python 2.7 and 3.5. Which Python version are you compiling against? (in your `yarp/build/`: `grep -P '^PYTHON_' CMakeCache.txt | grep -v INTERNAL`)

PeterBowman, 2017-09-09:
Looks like a SWIG issue instead (Trusty is stuck at 2.0.11, Xenial moved on to 3.0.8 ([ref](https://packages.ubuntu.com/search?keywords=swig&searchon=names))), see this [SO question](https://stackoverflow.com/q/18213813). As stated above, the offending lines are:

* Portable.h#L31 at `v2.3.70`
https://github.com/robotology/yarp/blob/6fafdd7ef80ec7dda33606288c413d89e45be8d5/src/libYARP_OS/include/yarp/os/Portable.h#L31

* Contact.h#L83 at `devel`:
https://github.com/robotology/yarp/blob/204093927be297847ec230edce990929f3f8b1b9/src/libYARP_OS/include/yarp/os/Contact.h#L83

...which use newer C++11 features (I guess that `YARP_OVERRIDE` resolves to the `override` specifier?). In fact, I don't think SWIG 2.0 supports them (compare [2.0 docs](http://www.swig.org/Doc2.0/SWIGPlus.html) vs [the dedicated C++11 section at 3.0](http://www.swig.org/Doc3.0/CPlusPlus11.html)).

PeterBowman, 2017-09-09:
And all this boils down to Ubuntu Trusty (14.04) being no longer supported by YARP (robotology/QA/issues/173). @Kuronogard you may still get positive results on 2.3.68.1. For 2.3.70 onwards, you should upgrade to Xenial.

traversaro, 2017-09-09:
Thanks for investigating the problem @PeterBowman . 
I think in YARP we should enforce the SWIG minimum required version at the CMake level, to provide a clear error in case any user tried to use an unsupported SWIG .

Kuronogard, 2017-09-09:
I think I'll keep going with an older version, then. Thanks for the help.

drdanz, 2017-09-14:
swig 3 (3.0.2) is available for Ubuntu 14.04 in the backports repository (see https://launchpad.net/ubuntu/+source/swig). You can try that one...

traversaro, 2017-10-27:
Fixed by https://github.com/robotology/yarp/pull/1350 .

closed 2017-10-27 by traversaro