robotology/yarp #1342: Publisher object does not work after RpcClient.open() call

Kuronogard, 2017-09-13:
I am writing a program to subscribe to ROS topics and call ROS services, however, the ports connected to topics don't seem to work if there is any opened RpcClient object.

My system is as follows:
 - OS: Ubuntu 14.04
 - YARP: 2.7.30
 - ROS: Indigo

I've tested this issue with a YARP program to connect to turtlesim's services and topics, and a ROS program to compare results.
[test_services.tar.gz](https://github.com/robotology/yarp/files/1299214/test_services.tar.gz). Build with `mkdir build; cd build; cmake ..; make`
[test_service_ros.tar.gz](https://github.com/robotology/yarp/files/1299216/test_service_ros.tar.gz) Built with `catkin_make` within a catkin workspace.

aerydna, 2018-08-20:
the test does not compile anymore. trying to replicate the same situation has not raised any problem on the current YARP (3.0.1).

closed 2018-08-20 by aerydna

Nicogene, 2018-08-27:
Marking fixed for now, it could be nice to discover what actually fixed this bug.
Bisecting could be the way.