robotology/yarp #1351: IPositionDirect::setPositions(const int, const int *, double *) should be setPositions(const int, const int *, const double *)

traversaro, 2017-09-20:
`IPositionDirect::setPositions(const int n_joint, const int *joints, double * refs)` is never meant to modify the  `refs` buffer, so its signature should actually be `IPositionDirect::setPositions(const int n_joint, const int *joints, `**`const`**` double * refs)`.

This is indeed the signature of other similar three parameters set methods, such as:
* `virtual bool positionMove(const int n_joint, const int *joints, const double *refs)`
* `virtual bool relativeMove(const int n_joint, const int *joints, const double *deltas)`
* `virtual bool velocityMove(const int n_joint, const int *joints, const double *spds)`

damn1, 2017-11-20:
As discussed, this issue seems not to be solvable without compatibility problems. Should it be a "won't fix"?

traversaro, 2017-11-20:
Unfortunately I am afraid this is the only option, unless we change the name of the methods (not desirable as well because it is too destructive).

drdanz, 2017-11-21:
:-1:
At some point we will have to break something...

drdanz, 2018-03-11:
The option described [here](https://stackoverflow.com/questions/49011990/) seems to be the less invasive option. Still we cannot get a build time warning when overriding the function, but we can add a runtime warning instead.

lornat75, 2018-03-23:
looks like a mistake while writing the function :(

drdanz, 2018-05-28:
Fixed in #1704

PR #1704 (2018-05-28): Const correctness
It fixes #1351 
It fixes #1616  
It fixes #1621  
It fixes #1617 
It fixes #1610 

`demoRedBall` successfully tested.
commit 5bf09a0f (2018-05-28): YARP_dev: fix const correctness of `IPositionDirect::setPositions`.

It fixes #1351
closed 2018-05-28 by drdanz