robotology/yarp #1357: On Windows "yarpdev --list" command results in segmentation fault.

damn1, 2017-09-21:
The program is blocked while trying to split the text describing devices:
```
Device "test_motor", C++ class yarp::dev::TestMotor, wrapped by "controlboard"
Device "remote_grabber", C++ class yarp::dev::RemoteFrameGrabber, wrapped by "grabber"
Device "grabber", C++ class yarp::dev::ServerFrameGrabber, is a network wrapper.
...
```
into lines with single device. It stops on the invocation of  getline function from ConstString class:
```
    inline std::istream& getline(std::istream& is, yarp::os::ConstString& str, char delim) {
        return getline(is, (std::string&)str, delim);
    }
```
where `is` seems to be void.

barbalberto, 2017-10-20:
Any update on this issue?

damn1, 2017-10-20:
I searched about this the day we discovered it, but in the end I was not able to solve the problem, it seems to be something non trivial. If you want we can discuss the details about this issue f2f.

drdanz, 2017-10-24:
On which Visual Studio version have you tried this?

damn1, 2017-10-24:
VS2015, sorry I didn't write it in the description of the issue. I remember @aerydna also tested it.

randaz81, 2017-10-24:
Also VS2013 Ultimate

francesco-romano, 2017-10-25:
@drdanz in `ConstString` there is the cast defined:
`operator std::string()`.

PR #1429 (2017-10-26): Bugfix for yarpdev on Windows
Fix `yarp::os::ConstString::getline(...)` to work fine with Windows.
It fixes #1357.
closed 2017-10-26 by drdanz