robotology/yarp #1358: RPATH error configuring yarp

diegoferigo, 2017-09-22:
Configuring yarp with the following options:

```
cmake \
      -DCMAKE_BUILD_TYPE=Debug \
      -DCMAKE_INSTALL_PREFIX=/iit/local \
      -DCREATE_GUIS=ON \
      -DCREATE_LIB_MATH=ON \
      ..
```

I get this RPATH error:

```
CMake Error at /iit/local/share/YCM/modules/AddInstallRPATHSupport.cmake:137 (file):
  file RELATIVE_PATH must be passed a full path to the directory:
  INSTALL_NAME_DIR
Call Stack (most recent call first):
  cmake/YarpOptions.cmake:359 (add_install_rpath_support)
  CMakeLists.txt:63 (include)
```

cc @francesco-romano @drdanz @claudiofantacci

francesco-romano, 2017-09-22:
You have an old version of YCM (or YCM has not been synched to YARP yet).
Remove YCM from the path and retry

diegoferigo, 2017-09-22:
This error happens if I install YCM separately (from `devel` in my case, 0.5.0+git version) before YARP. If I leave YARP handle YCM it works flawlessly :+1: 

It looks like after the most recent YCM commits, a synchronization from the YARP side is still pending.

drdanz, 2017-09-25:
It should be in sync... Are you using YCM `devel` branch?

diegoferigo, 2017-09-26:
I rebuilt everything from scratch in a clean environment. So far so good, thanks! ;) Probably there were some leftovers in the system :sweat_smile: Closing

closed 2017-09-26 by diegoferigo