robotology/yarp #1365: import error: No module named yarp

muhammadbilalababsi, 2017-09-25:
Hi, 
    I am working first time on ubuntu 16.04 xenial. I have already installed yarp , SWIG and iCub_SIM. I have two operating systems on my pc(windows 10 and ubuntu 16.04) and i also have virtual machine for running ubuntu on windows. On virtual machine my code is running correctly but when i run it on other operating system(ubuntu) it gives the following error.
                                    "import error: no module named yarp"
Can anyone please tell me what is wrong with it? i also set the environment variables as specified on the installation guide and reinstalled yarp and reinstalled ubuntu as well but problem seems to be the same. thanks

traversaro, 2017-09-25:
Hi @muhammadbilalababsi , 
  what are you trying to run when you get the reported error? 
A Python script that uses YARP bindings?

muhammadbilalababsi, 2017-09-25:
yes

muhammadbilalababsi, 2017-09-25:
#!/usr/bin/python

import numpy
import yarp
import matplotlib.pylab
import time

yarp.Network.init()
if yarp.Network.checkNetwork() != True:
	print "[error] Please try running yarp server"
	quit()

# prepare a property object
props = yarp.Property()
props.put("device","remote_controlboard")
props.put("local","/client/head")
props.put("remote","/icubSim/head")

# create remote driver
DeadDriver = yarp.PolyDriver(props)

#query motor control interfaces
iPos = DeadDriver.viewIPositionControl()
iVel = DeadDriver.viewIVelocityControl()
iEnc = DeadDriver.viewIEncoders()

#retrieve number of joints
jnts=iPos.getAxes()

print 'Controlling', jnts, 'joints'
 
# read encoders
encs=yarp.Vector(jnts)
iEnc.getEncoders(encs.data())
 
home=yarp.Vector(jnts, encs.data())

########

input_port_right_eye = yarp.Port()
input_port_right_eye.open("/eyes/right")
if yarp.Network.connect("/icubSim/cam/right", "/eyes/right") != True:
	print "[error] Could not connect input_port_right_eye"
	quit()

input_port_left_eye = yarp.Port()
input_port_left_eye.open("/eyes/left")
if yarp.Network.connect("/icubSim/cam/left", "/eyes/left") != True:
	print "[error] Could not connect input_port_left_eye"
	quit()

output_port_right_eye = yarp.Port()
output_port_right_eye.open("/view/right")
if yarp.Network.connect("/eyes/right", "/view/right") != True:
	print "[error] Could not connect input_port_right_eye"
	quit()
	

# Create numpy array to receive the image and the YARP image wrapped around it
left_eye_img_array = numpy.zeros((240, 320, 3), numpy.uint8) 
left_eye_yarp_image = yarp.ImageRgb()
left_eye_yarp_image.resize(320, 240)

right_eye_img_array = numpy.zeros((240, 320, 3), numpy.uint8) 
right_eye_yarp_image = yarp.ImageRgb()
right_eye_yarp_image.resize(320, 240)

left_eye_yarp_image.setExternal(left_eye_img_array, left_eye_img_array.shape[1], left_eye_img_array.shape[0])
right_eye_yarp_image.setExternal(right_eye_img_array, right_eye_img_array.shape[1], right_eye_img_array.shape[0])

#initialize a new tmp vector identical to encs
tmp=yarp.Vector(jnts)

input_port_left_eye.read(left_eye_yarp_image)
input_port_right_eye.read(right_eye_yarp_image)
	
input_port_left_eye.read(left_eye_yarp_image)
input_port_right_eye.read(right_eye_yarp_image)

input_port_left_eye.read(left_eye_yarp_image)
input_port_right_eye.read(right_eye_yarp_image)

for x in xrange(1, 11):

	print x 
	
	tmp.set(0, tmp.get(0)-3)
	tmp.set(1, tmp.get(1)-1)
	tmp.set(2, tmp.get(2)-1)
	tmp.set(3, tmp.get(3)-1)
	tmp.set(4, 0)
	tmp.set(5, 0)
	
	iPos.positionMove(tmp.data())
	
	# Read the data from the port into the image
	input_port_left_eye.read(left_eye_yarp_image)
	input_port_right_eye.read(right_eye_yarp_image)
		

	if left_eye_yarp_image.getRawImage().__long__() <> left_eye_img_array.__array_interface__['data'][0]:
		print "read() reallocated my left_eye_yarp_image!"
	
	if right_eye_yarp_image.getRawImage().__long__() <> right_eye_img_array.__array_interface__['data'][0]:
		print "read() reallocated my right_eye_yarp_image!"

	plot_image = numpy.concatenate((left_eye_img_array, right_eye_img_array), axis=1)
	matplotlib.pylab.imshow(plot_image)
	matplotlib.pylab.show(block=False)
	#matplotlib.pylab.show()
	matplotlib.pylab.pause(0.05)
	#yarp.Time.delay(1.5)

# display the image that has been read
#matplotlib.pylab.imshow(left_eye_img_array)
#matplotlib.pylab.show(block=False)
#matplotlib.pylab.imshow(right_eye_img_array)
#matplotlib.pylab.show(block=False)

# Cleanup
input_port_left_eye.close()
input_port_right_eye.close()

yarp.Network.fini();

traversaro, 2017-09-25:
Then it seems an issue related to where did you installed the YARP Python bindings and how did you set [`PYTHONPATH`](https://docs.python.org/3/using/cmdline.html#envvar-PYTHONPATH). 
Are you using the Python bindings from the build folder of YARP or from the installation folder? Which instructions are you using to compile and install the Python bindings?

muhammadbilalababsi, 2017-09-25:
i used the following commands to set the environment variables.
export YARP_DIR=$muhammad/Downloads
export YARP_DATA_DIRS= $YARP_DIR/share/yarp
export YARP_DIR/bin=PATH
export YARP_DIR/lib=LD_LIBRARY_PATH

and further both of them i.e yarp and python script are in same location

muhammadbilalababsi, 2017-09-25:
for installing python bindings i used this command 
sudo apt-get install swig

muhammadbilalababsi, 2017-09-25:
for compilling pythonn yarp bindings i used these commands
cd $YARP_ROOT/bindings
mkdir build
cd build
ccmake ..
make
sudo make install

export PYTHONPATH=$PYTHONPATH:/path/to/bindings/build

traversaro, 2017-09-28:
In the command
~~~
export PYTHONPATH=$PYTHONPATH:/path/to/bindings/build
~~~
I guess that the `/path/to/bindings/build` should be substituted with the actual location of the build directory of the bindings, in your case I imagine `$YARP_ROOT/bindings/build`. 

However, to enable us to help you, can you post the link to the instructions that you are following? Thanks a lot!

drdanz, 2017-10-17:
Closing this, please reopen if you still have issues

closed 2017-10-17 by drdanz

muhammadbilalababsi, 2017-10-17:
Thank you Mr. @traversaro  for your help, i got my problem resolved. i was using these simple steps to install yarp from source.    http://www.yarp.it/install_yarp_linux.html