robotology/yarp #1373: Document/clarify that multiple yarp::os::Node objects are allowed for process

traversaro, 2017-09-27:
The [`yarp::os::Node`](http://www.yarp.it/classyarp_1_1os_1_1Node.html) class is a YARP class useful to interact with the [ROS 1 Master](http://wiki.ros.org/Master) and publish/subscribe ROS 1 topics without the need to link any ROS client library such as [roscpp](http://wiki.ros.org/roscpp).

In the [ROS 1 computation graph](http://wiki.ros.org/ROS/Concepts#ROS_Computation_Graph_Level), the [Node](http://wiki.ros.org/Nodes) identifies a single [process](https://en.wikipedia.org/wiki/Process_(computing)). However, using the `yarp::os::Node` is possible to create multiple `Node`s for each process. It is not clear (for some of us) if this is allowed or it can create problem with ROS 1 Master.

* [ ] Document the class and its methods with Doxygen

cc @drdanz @randaz81 @aerydna and in particular @barbalberto that worked a lot on ROS/YARP integration

traversaro, 2017-09-27:
An example of use of multiple nodes in the same process that is causing a crash is attached. 

[testPublisher3.zip](https://github.com/robotology/yarp/files/1336614/testPublisher3.zip)

damn1, 2018-07-11:
I've tested this snippet of code with YARP 2.3.72, then ported it to YARP 3.0.0 and retested. I'm not able to reproduce this problem n a debugging context. Do you remember what kind of crash should it cause? 
Here is a gist with the same test that can build with YARP 3.0.0:
https://gist.github.com/damn1/6355b4c952bcba4451230e2c2ef52017

Maybe this issue could be unified with #1774 depending on the crash it was causing?

traversaro, 2018-07-11:
I do not remember correctly how the details on how I run those tests, but based on the date of the issue there are some aspects that perhaps you did not replicated on your setup: 
* The `yarpserver` was running on a machine different from the one on which I was running the test (possibly I was communicating with it over WiFi) 
* `roscore` was running, and the `yarpserver` was launched with the `--ros` option (and perhaps also the yarpserver and the roscore were communicating over WiFi.

traversaro, 2018-07-11:
I remember I also used `rqt_graph` and I obtained strange results, wondering if the multuple node for a single process was creating problems with the ROS 1 `roscore`.

traversaro, 2018-07-12:
@damn1 This issue could be related: https://github.com/robotology/yarp/issues/1369 .