robotology/yarp #1377: yarp::os::Time::useNetworkClock should take in input a timeout parameter

traversaro, 2017-09-28:
I am debugging a strange behavior of the `yarp::os::Time::useNetworkClock` in which the network clock is never valid even if the specified port exists, and the first problem I am experiencing is that the useNetworkClock method hangs forever. Similar to the checkNetwork method, the useNetworkClock should have an optional timeout parameter, to return after some time even if the network clock never published anything.

barbalberto, 2017-10-04:
Personally I do not agree with this option for 2 reasons, see after for a different solution.

1) For the checkNetwork, not being able to find the yarpserver is a buggy situation, while for the clock not receiving an input for an unknown long time is a perfectly normal condition.
For example if I start an executable when Gazebo is in pause, maybe I simply want it to wait until I continue the simulation.
Another example is a user that want to start all clock-dependant applications first, and then the clock as last application to force all other executable to a start at the same time.
Adding a timeout will force the user to a specific workflow (first the clock, then other app) and I'm not sure it's worth it.

2) Clock initializaztion, in general, cannot fail because the code cannot run without it. So what shall YARP do when the timeout expires? The only options are:
 - Fallback to systemClock: This is no-go (evil), because it means not respecting user specifications, and you'll have different executable silently using different clocks depending on when the network clock producer is started or on system responsiveness when multiple executable are started at the same time.
 - Assert: This is doable, but I think too rude. As stated above, in my opinion this is not an error condition for the clock consumer (it may be an error condition for the clock producer), so why shall **_I_** die? 

In general, I prefer to avoid timeouts as much as possible, expecially here in the `Clock` functions.

If the problem is "I have something stuck in the initialization of NetworkClock and I'm not able to close it" brand new `yarp clock` companion comes to the rescue. It opens a `/clock` port and broadcast the time, this is handy for debugging and unlock stuck code.

I'd rather add an `abort` command to the companion forcing an assert to the connected clients, because it keeps the deterministic behaviour of the whole system.

traversaro, 2017-10-04:
> If the problem is "I have something stuck in the initialization of NetworkClock and I'm not able to close it" brand new `yarp clock` companion comes to the rescue. It opens a `/clock` port and broadcast the time, this is handy for debugging and unlock stuck code.

Not related to this issue, but in my case the problem was that the clock was up, but for some reason the connection was not established and so `yarp clock` was not really useful. 

> In general, I prefer to avoid timeouts as much as possible, especially here in the `Clock` functions.

I agree about "hidden" timeout and all the related problems (what to do in case of the timeout expires?). It was not probably clear in the original issue,  but I was thinking about something like a new function:
~~~
static bool useNetworkClock (const ConstString &clock, double timeout)
~~~
In case the timeout expires, the function simply returns false.

barbalberto, 2017-10-04:
> Not related to this issue, but in my case the problem was that the clock was up, but for some reason the connection was not established and so yarp clock was not really useful.

`yarp clock` cannot help debugging a non working clock producer, but will unlock the clients from the network clock initialization.

> In case the timeout expires, the function simply returns false.

I understand what you mean, the whole point 2 in the previous comment was about the fact that function `useNetworkClock` _**cannot return false**_ in any case, because it will leave yarp without any clock set. My first implementation was with returning a `bool`, but the reverted to `void` because that situation cannot be handled in a safe way.

For example the code:
```
useNetworkClock("/clock", 5.0);   // timeout happens
Time::now();   --> what shall it do?
```
Clock is too critical to let the user the burden of checking if initialization went right.
The only doable thing is that is timeout expires, the function calls an Assert, but cannot go further.

traversaro, 2017-10-04:
Now I got it. As discussed f2f, the proper way to proceed for a use case of this kind is to have a call to `exists` before any call to `useNetworkClock` .

closed 2017-10-04 by traversaro