robotology/yarp #1401: The "Check All" function of the Cluster Management Window in yarpmanager is not working

S-Dafarra, 2017-10-11:
If we launch ``yarprun`` on the robot's head (using ``yarpmanager``) and then we shut the head down, the status in the ``Cluster Management Window`` remains green, even if we press ``Check All``.

@Nicogene @gabrielenava

Nicogene, 2017-10-11:
The nameserver runs on the head?

francesco-romano, 2017-10-11:
@S-Dafarra have you tried a `yarp clean`?
Usually `yarprun` creates a port, and that port is checked by the manager

Nicogene, 2017-10-11:
@francesco-romano  is right, if you do not perform yarp clean, the port with the `yarprun` property is still registered in the nameserver, so for the cluster is still there.

S-Dafarra, 2017-10-11:
> The nameserver runs on the head?

Nope
> @S-Dafarra have you tried a yarp clean?

Nope :grin:. But it was not necessary with the old cluster if I remember well.

Nicogene, 2017-10-11:
The old cluster did "Network::exists()" so I think that it is the same case. If the cluster continues to see "green" the head also after the `yarp clean` then there is a bug, otherwise I think it is correct as it is. Eventually I can add a `yarp clean` button in `yarpmanager` :wink:

S-Dafarra, 2017-10-11:
I see, thanks. If we can then talk about enhancements, 
- is it possible to do the check automatically? I.e. can we comunicate with the specific port to see whether it is still active (similar to what ``yarp clean`` does)?
- can ``Stop Selected`` or ``Kill Selected`` make it "red" in case of no response. Otherwise one may be wondering why ``yarprun`` refuses to close while instead it is already dead :grin:

Nicogene, 2017-10-12:
Ok for the first point, I don't think it is a big problem to implement it.
About the second point `Stop Selected` `Kill Selected`  are simply : `ssh user@node killall yarprun` and `ssh user@node killall -9 yarprun`, if the node is shut down it doesn't do nothing :sweat_smile: 
Thanks for the feedback!

S-Dafarra, 2017-10-12:
> Ok for the first point, I don't think it is a big problem to implement it.

 Great! :smiley: 
>if the node is shut down it doesn't do nothing 

Maybe if the first point can be achieved, after the ``ssh user@node killall -9 yarprun`` command, the same check could be performed to turn the state to "red". 
>Thanks for the feedback!

Thank you for supporting :wink:

Nicogene, 2017-10-18:
Fixed by #1414

PR #1414 (2017-10-18): Improvement yarpmanager
After #1407 , it has been possible to introduces some new nice features:
- `yarp clean` button: before actually performing it, it asks which timeout do you want to use. It fixes #1401 
- `yarp name list` button:  "Ports" has been added in the entities tree, it is populated by this button.
:warning:  at the startup `yarpmanager` executes a `yarp name list` command.

[image]

Moreover it fixes #1089 .
Please review code.

demoRedBall TESTED.
closed 2017-10-18 by Nicogene