robotology/yarp #1404: arbitrary ip adress with yarpserver --read (ubuntu)

aerydna, 2017-10-12:
in ubuntu, despite configuring yarp on a arbitrary unexistent ip adress, yarp server with the `--read` param print on the terminal a (obviously false) correct start.

claudiofantacci, 2017-10-12:
I don't understand this issue. Could you please rephrase the problem and provide an example?
In particular, I don't understande
> `yarpserver` with the --read param print on the terminal a (obviously false) correct start.

drdanz, 2017-11-21:
Fixed in #1404

randaz81, 2018-04-11:
There exists a very limited and specific use case in which the user wants to start the yarp server using the --read option even if the specified ip address is not currently available (for example when using a wi-fi)
#1457 forbids it. We need to reformulate this fix.

valegagge, 2018-04-11:
@randaz81 just out of curiosity....why when we use wifi we don't know the ip address?

randaz81, 2018-06-04:
Additional use cases include yarp usage with DMZ and NAT.
commit 7ec99a6c73729 has been reverted: the original behavior of yarp --read has been restored.

PR #1457 (2017-11-21): Fixed issue #1404 about 'yarp server --read command'
If in config file there was a nonexistent IP address then the server started anyway.
Fixes #1404.

This patch should fix this problem: the application warns the user that it cannot start because the IP written in configuration file doesn't exist.
closed 2018-06-04 by randaz81