robotology/yarp #1415: yarp port randomly inserting wrong the scientific notation

fjandrad, 2017-10-17:
I launched yarpdatadumper from yarpmanager connected to the stateExt ports of the icub.
 I read following the format of the file and encounter some troubles, after a long search I found that sometimes I find the following error in the formatting

> 18967 1508247291.781955 (-0.0274659 -0.0219727 0.060425 0.0329591 0.0 0.0) [ok] (0.0 0.0 0.0 0.0 0.0 0.0) [ok] (0.0 0.0 0.0 0.0 0.0 0.0) [ok] (-64.5613 85.787 -701.479 115.406 0.0 0.0) [ok] (0.0 0.0 0.0 0.0 0.0 0.0) [ok] (0.0 0.0 0.0 0.0 0.0 0.0) [ok] (-0.110125 -0.009206 -7.3e-05 0.0 0.0 0.0) [ok] (2.97619 -5.11905 -0.357143 8.18452 6.66667 -28.0952) [ok] (7.0 -30.0 0.0 30.0 15.0 -122.0) [ok] (7565168 7565168 7565168 7565168 7565168 7565168) [ok] (1718187123 1718187123 1718187123 1718187123 1718187123 1718187123) [ok]
> 
> 18968 1508247291.791997 (-0.0274659 -0.0219727 0.060425 0.0329591 0.0 0.0) [ok] (0.0 0.0 0.0 0.0 0.0 0.0) [ok] (0.0 0.0 0.0 0.0 0.0 0.0) [ok] (-64.5613 85.787 -701.479 115.406 0.0 0.0) [ok] (0.0 0.0 0.0 0.0 0.0 0.0) [ok] (0.0 0.0 0.0 0.0 0.0 0.0) [ok] (-0.121318 -0.009059 **-1e-06.0** 0.0 0.0 0.0) [ok] (2.82738 -4.94048 -0.357143 8.18452 6.66667 -28.0952) [ok] (3.0 -30.0 -7.0 15.0 26.0 -110.0) [ok] (7565168 7565168 7565168 7565168 7565168 7565168) [ok] (1718187123 1718187123 1718187123 1718187123 1718187123 1718187123) [ok]
> 

I am not sure if it only happens when the number is for some reason I have only found it happens with e-06.0 and e-05.0 and in some stateExt files

fjandrad, 2017-10-17:
@traversaro  do not know who to ask to look into this xD
@gabrielenava I finally found it!!!!!

pattacini, 2017-10-17:
Maybe related to #1400 ?

traversaro, 2017-10-18:
Fascinating. 
Given https://github.com/robotology/yarp/blob/9a395f6e41e01d8d350b6a4e5e2b2f6c6f913b74/src/libYARP_dev/src/devices/msgs/stateExt.thrift , it appears that that those numbers are torques. Did you experience this only on torques or also on other fields in `stateExt:o`? 

cc @barbalberto

barbalberto, 2017-10-18:
Why wrong? Can you better explain what do you mean by wrong?

francesco-romano, 2017-10-18:
@barbalberto the decimal part should be before the exponential:
```
-1.0e-06
```

closed 2017-10-18 by drdanz

drdanz, 2017-10-18:
Should be fixed in latest master. YARP was appending ".0" when a decimal point was not found in the number to ensure that it would be interpreted as a double. Now the decimal point is appended only when the decimal point and the exponential are not found, therefore it will be printed as `-1e-06`

barbalberto, 2017-10-18:
yes, I didn't notice at the first sight. Fixed.

fjandrad, 2017-10-20:
I'm not sure at the moment but I think it was always on the torques. Though I can not confirmed since I replaced manually all the files I had with this error since I wanted to calibrate the sensors ASAP.
I have not tried collecting more data and see if it happens but since it is fixed I think is not relevant anymore.

commit ffbee32e (2017-10-18): OS/BottleImpl: Do not append ".0" when using exponential notation

Fixes #1415