robotology/yarp #1423: Compile without gstreamer

diegoferigo, 2017-10-24:
Building YARP with these options:
```
cmake -DCREATE_GUIS=ON \
      -DCREATE_LIB_MATH=ON \
      -DSKIP_ACE=ON \
      ..
```
If the libraries reported in https://github.com/robotology/yarp/pull/1346#issuecomment-334801437 are not installed, the build fails with the following error:
```
[...]
--  --- plugin yarpcar_priority: disabled (dependencies unsatisfied)
--  --- plugin yarpcar_portmonitor: disabled (dependencies unsatisfied)
--  --- plugin yarpcar_depthimage: disabled (dependencies unsatisfied)
--  --- plugin yarpcar_zfp: disabled (dependencies unsatisfied)
--  --- plugin yarpcar_h264: disabled (dependencies unsatisfied)
-- ending plugin library: yarpcar
-- yarpdatadumper: OpenCV not selected, keep on building...
-- yarpdataplayer: OpenCV not selected, keep on building...
-- In-tree includes: /iit/sources/yarp/build/src/libYARP_conf/include;/iit/sources/yarp/src/libYARP_OS/include;/iit/sources/yarp/src/libYARP_sig/include;/iit/sources/yarp/src/libYARP_eigen/include;/iit/sources/yarp/src/libYARP_gsl/include;/iit/sources/yarp/src/libYARP_math/include;/iit/sources/yarp/src/libYARP_dev/include;/iit/sources/yarp/src/libYARP_name/include;/iit/sources/yarp/src/libYARP_manager/include;/iit/sources/yarp/src/libYARP_profiler/include;/iit/sources/yarp/src/libYARP_logger/include
-- YARP libraries: YARP_priv_xmlrpcpp;YARP_conf;YARP_OS;YARP_sig;YARP_eigen;YARP_gsl;YARP_math;YARP_dev;YARP_name;YARP_serversql;YARP_manager;YARP_profiler;YARP_logger;YARP_init
-- Configuring done
-- Generating done
-- Build files have been written to: /iit/sources/yarp/build
CMake Error: Parse error in cache file /iit/sources/yarp/build/CMakeCache.txt on line 536. Offending entry: Perhaps you should add the directory containing `gstreamer-1.0.pc'
CMake Error: Parse error in cache file /iit/sources/yarp/build/CMakeCache.txt on line 537. Offending entry: to the PKG_CONFIG_PATH environment variable
CMake Error: Parse error in cache file /iit/sources/yarp/build/CMakeCache.txt on line 538. Offending entry: No package 'gstreamer-1.0' found
-- YARP Version: 2.3.71 (2.3.71+20171024.15+gitf29bd015b)
-- Using:
--   CMake: 3.7.2 (/usr/bin/cmake)
--   C Compiler: GNU 6.3.0 (/usr/bin/cc)
--   CXX Compiler: GNU 6.3.0 (/usr/bin/c++)
-- Detecting required libraries
-- CMake modules directory: /iit/local/share/YCM/modules;/iit/local/share/YCM/find-modules;/iit/local/share/YCM/build-modules;/iit/local/share/YCM/cmake-next/Modules;/iit/local/share/YCM/3rdparty;/iit/local/share/YCM/deprecated;/iit/sources/yarp/cmake
-- Checking GNUCXX version 3/4 to determine  OpenCV /opt/net/ path
-- - DBG OPENCV_INCLUDE_DIR_CVCAM=OPENCV_INCLUDE_DIR_CVCAM-NOTFOUND
-- - DBG OPENCV_INCLUDE_DIR_CVCAM=OPENCV_INCLUDE_DIR_CVCAM-NOTFOUND
-- OPENCV_LIBRARY not found turning off OPENCV_FOUND
-- OPENCV_CVAUX_LIBRARY not found turning off OPENCV_FOUND
-- OPENCV library or headers not found. Please search manually or set env. variable OPENCV_ROOT to guide search.
-- Checking for module 'gstreamer-1.0'
--   No package 'gstreamer-1.0' found
-- Could NOT find GStreamer (missing:  GSTREAMER_LIBRARY GSTREAMER_INCLUDE_DIR GSTREAMER_VERSION_COMPATIBLE) (Required is at least version "1.4")
-- Checking for module 'gstreamer-plugins-base-1.0'
--   No package 'gstreamer-plugins-base-1.0' found
-- Checking for module 'gstreamer-app-1.0'
--   No package 'gstreamer-app-1.0' found
-- Could NOT find GSTREAMER_APP_LIBRARY
-- Could NOT find GSTREAMER_APP_INCLUDE_DIR
-- I have found the following libraries:
--  +++ YCM (>= 0.5.1): found 0.5.1.20171006.2+git6f29deb (/iit/local/share/YCM/cmake)
--  --- ACE: disabled
--  --- RTF (>= 1.3.3): not found
--  +++ SQLite: found 3.16.2 (/usr/lib/x86_64-linux-gnu/libsqlite3.so)
--  +++ Eigen3: found (/usr/include/eigen3)
--  +++ TinyXML (>= 2.6): found 2.6.2 (/usr/lib/x86_64-linux-gnu/libtinyxml.so)
--  +++ Qt5: found 5.7.1 (/usr/lib/x86_64-linux-gnu/cmake/Qt5)
--  +++ QCustomPlot: found (/usr/lib/x86_64-linux-gnu/libqcustomplot.so)
--  --- Graphviz: disabled
--  +++ Libedit: found 3.1 (/usr/lib/x86_64-linux-gnu/libedit.so)
--  --- SWIG: disabled
--  --- OpenCV: not found
--  --- Lua: not found
--  --- LibOVR (>= 1.3): not found
--  --- GLFW3: not found
--  --- GLEW: not found
--  --- Libdc1394 (>= 2.0): not found
--  +++ JPEG: found (/usr/lib/x86_64-linux-gnu/libjpeg.so)
--  +++ MPI: found (/usr/lib/x86_64-linux-gnu/openmpi/lib/libmpi_cxx.so)
--  --- FTDI: not found
--  --- CUDA: not found
[... 54 lines cut ...]
--  --- plugin yarpcar_human: disabled (dependencies unsatisfied)
--  --- plugin yarpcar_mpi: disabled (dependencies unsatisfied)
--  --- plugin yarpcar_mpibcast: disabled (dependencies unsatisfied)
--  --- plugin yarpcar_xmlrpc: disabled (dependencies unsatisfied)
--  --- plugin yarpcar_tcpros: disabled (dependencies unsatisfied)
--  --- plugin yarpcar_rossrv: disabled (dependencies unsatisfied)
--  --- plugin yarpcar_mjpeg: disabled (dependencies unsatisfied)
--  --- plugin yarpcar_bayer: disabled (dependencies unsatisfied)
--  --- plugin yarpcar_priority: disabled (dependencies unsatisfied)
--  --- plugin yarpcar_portmonitor: disabled (dependencies unsatisfied)
--  --- plugin yarpcar_depthimage: disabled (dependencies unsatisfied)
--  --- plugin yarpcar_zfp: disabled (dependencies unsatisfied)
--  --- plugin yarpcar_h264: disabled (dependencies unsatisfied)
-- ending plugin library: yarpcar
-- yarpdatadumper: OpenCV not selected, keep on building...
-- yarpdataplayer: OpenCV not selected, keep on building...
-- In-tree includes: /iit/sources/yarp/build/src/libYARP_conf/include;/iit/sources/yarp/src/libYARP_OS/include;/iit/sources/yarp/src/libYARP_sig/include;/iit/sources/yarp/src/libYARP_eigen/include;/iit/sources/yarp/src/libYARP_gsl/include;/iit/sources/yarp/src/libYARP_math/include;/iit/sources/yarp/src/libYARP_dev/include;/iit/sources/yarp/src/libYARP_name/include;/iit/sources/yarp/src/libYARP_manager/include;/iit/sources/yarp/src/libYARP_profiler/include;/iit/sources/yarp/src/libYARP_logger/include
-- YARP libraries: YARP_priv_xmlrpcpp;YARP_conf;YARP_OS;YARP_sig;YARP_eigen;YARP_gsl;YARP_math;YARP_dev;YARP_name;YARP_serversql;YARP_manager;YARP_profiler;YARP_logger;YARP_init
-- Configuring incomplete, errors occurred!
See also "/iit/sources/yarp/build/CMakeFiles/CMakeOutput.log".
See also "/iit/sources/yarp/build/CMakeFiles/CMakeError.log".
make: *** [cmake_check_build_system] Error 1
Makefile:834: recipe for target 'cmake_check_build_system' failed
```
Are these libraries required by all the installations now?

Nicogene, 2017-10-24:
They are required if you want to compile `h264` carrier, your problem seem related to the CmakeCache.. Have you tried to delete it ? Or to do `rm -rf *` in the build directory?

diegoferigo, 2017-10-24:
@Nicogene thanks for your comment, this happens with a clean build environment in my docker setup (hence it is fully reproducible).

I dug more into my build process and I figured out that the configuring of the project succeeds, is the second configuring which happens as first step of the building project that fails. I updated the first post with further details.

Nicogene, 2017-10-24:
But it is strange that it performs the second configuration if the first succeeded :thinking: by the way I don't think it is related to gstreamer or `h264` because it is disabled :
`--- plugin yarpcar_h264: disabled (dependencies unsatisfied)`

drdanz, 2017-10-24:
@diegoferigo Can you try with YCM disabled?

diegoferigo, 2017-10-24:
Yes I agree it is strange. I don't mind having the `gstreamer` packages installed in my system and I'm living fine with them, I just wanted to report this (low importance) issue for future reference. Feel free to pass at my desk ;)

@drdanz 
With:

`cmake -DCREATE_GUIS=ON -DCREATE_LIB_MATH=ON -DSKIP_ACE=ON  -DYARP_USE_YCM=OFF ..`

The second configuring doesn't happen, and all works fine.

diegoferigo, 2017-10-24:
~The second configuring doesn't happen, and all works fine.~
Edit: cleaned the build and it fails with the same error

drdanz, 2017-10-24:
Can you please try also `CMAKE_DISABLE_FIND_PACKAGE_YCM=ON`?

diegoferigo, 2017-10-24:
With that options, the second configuring which happens executing `make` doesn't happen, all is fine with that.

Nicogene, 2017-10-24:
@drdanz is it a YCM bug or yarp bug ? Shall we close this issue?

drdanz, 2017-10-24:
YCM bug, the Find files were updated in YARP, but not in YCM...
Just leave this open, I will fix it tomorrow...

Nicogene, 2017-11-16:
>Just leave this open, I will fix it tomorrow...
@drdanz :smile:

aerydna, 2017-11-17:
fixed by robotology/ycm@3784576

closed 2017-11-17 by aerydna