robotology/yarp #1428: Release of drdanz v1.38.0

Nicogene, 2017-10-25:
DRDANZ 1.38.0 (2017-10-24) Release Notes                                {#v1_38_0}
======================================

Important Changes
-----------------

### Libraries

* A lot of refactoring was performed in order to improve c++11 usage and to
  simplify porting to other platforms without ACE

#### `DANZ_family`

* Added new daughter `Marta`.

#### `DANZ_work`

* Added three new slaves : @Nicogene,  @aerydna, @damn1.

#### `DANZ_fun`

* Increased abilities with Kirby in Super Smash Bros.
* Added `Nintendo Switch`. 

@drdanz  is the **maintainer** that YARP deserves, but not the one it needs right now.

Happy BDAY  :tada: :tada:

aerydna, 2017-10-25:
this release looks almost good to me.. it seems something is missing

aerydna, 2017-10-25:
ah.. there is a pr still under test.. i think it will finish by October 27th

Nicogene, 2017-10-25:
This PR will be an **odissey** ....

closed 2017-10-25 by Nicogene

randaz81, 2017-10-26:
UH this PR has been already closed but I didn't have the chance to review it?!

drdanz, 2017-10-30:
Merged, thanks.