robotology/yarp #1433: Compilation failing on YARP devel on Ubuntu 16.04

traversaro, 2017-10-27:
Commit: aac0dc7188449dc2707f6062a5646639b5667793 . 
~~~
[ 28%] Building CXX object src/libYARP_sig/CMakeFiles/YARP_sig.dir/src/ImageCopy.cpp.o
In file included from /home/traversaro/src/something-superbuild/something/YARP/src/libYARP_sig/src/ImageCopy.cpp:8:0:
/home/traversaro/src/something-superbuild/something/YARP/src/libYARP_sig/include/yarp/sig/Image.h:593:69: error: specialization of ‘template<class T> virtual int yarp::sig::ImageOf<T>::getPixelCode() const’ in different namespace [-fpermissive]
 inline int yarp::sig::ImageOf<yarp::sig::PixelMono>::getPixelCode() const {
                                                                     ^
/home/traversaro/src/something-superbuild/something/YARP/src/libYARP_sig/include/yarp/sig/Image.h:563:17: error:   from definition of ‘template<class T> virtual int yarp::sig::ImageOf<T>::getPixelCode() const’ [-fpermissive]
     virtual int getPixelCode() const override;
                 ^
/home/traversaro/src/something-superbuild/something/YARP/src/libYARP_sig/include/yarp/sig/Image.h:598:71: error: specialization of ‘template<class T> virtual int yarp::sig::ImageOf<T>::getPixelCode() const’ in different namespace [-fpermissive]
 inline int yarp::sig::ImageOf<yarp::sig::PixelMono16>::getPixelCode() const {
                                                                       ^
/home/traversaro/src/something-superbuild/something/YARP/src/libYARP_sig/include/yarp/sig/Image.h:563:17: error:   from definition of ‘template<class T> virtual int yarp::sig::ImageOf<T>::getPixelCode() const’ [-fpermissive]
     virtual int getPixelCode() const override;
                 ^
/home/traversaro/src/something-superbuild/something/YARP/src/libYARP_sig/include/yarp/sig/Image.h:603:68: error: specialization of ‘template<class T> virtual int yarp::sig::ImageOf<T>::getPixelCode() const’ in different namespace [-fpermissive]
 inline int yarp::sig::ImageOf<yarp::sig::PixelRgb>::getPixelCode() const {
                                                                    ^
/home/traversaro/src/something-superbuild/something/YARP/src/libYARP_sig/include/yarp/sig/Image.h:563:17: error:   from definition of ‘template<class T> virtual int yarp::sig::ImageOf<T>::getPixelCode() const’ [-fpermissive]
     virtual int getPixelCode() const override;
                 ^
/home/traversaro/src/something-superbuild/something/YARP/src/libYARP_sig/include/yarp/sig/Image.h:608:69: error: specialization of ‘template<class T> virtual int yarp::sig::ImageOf<T>::getPixelCode() const’ in different namespace [-fpermissive]
 inline int yarp::sig::ImageOf<yarp::sig::PixelRgba>::getPixelCode() const {
                                                                     ^
/home/traversaro/src/something-superbuild/something/YARP/src/libYARP_sig/include/yarp/sig/Image.h:563:17: error:   from definition of ‘template<class T> virtual int yarp::sig::ImageOf<T>::getPixelCode() const’ [-fpermissive]
     virtual int getPixelCode() const override;
                 ^
/home/traversaro/src/something-superbuild/something/YARP/src/libYARP_sig/include/yarp/sig/Image.h:613:68: error: specialization of ‘template<class T> virtual int yarp::sig::ImageOf<T>::getPixelCode() const’ in different namespace [-fpermissive]
 inline int yarp::sig::ImageOf<yarp::sig::PixelHsv>::getPixelCode() const {
                                                                    ^
/home/traversaro/src/something-superbuild/something/YARP/src/libYARP_sig/include/yarp/sig/Image.h:563:17: error:   from definition of ‘template<class T> virtual int yarp::sig::ImageOf<T>::getPixelCode() const’ [-fpermissive]
     virtual int getPixelCode() const override;
                 ^
/home/traversaro/src/something-superbuild/something/YARP/src/libYARP_sig/include/yarp/sig/Image.h:618:68: error: specialization of ‘template<class T> virtual int yarp::sig::ImageOf<T>::getPixelCode() const’ in different namespace [-fpermissive]
 inline int yarp::sig::ImageOf<yarp::sig::PixelBgr>::getPixelCode() const {
                                                                    ^
/home/traversaro/src/something-superbuild/something/YARP/src/libYARP_sig/include/yarp/sig/Image.h:563:17: error:   from definition of ‘template<class T> virtual int yarp::sig::ImageOf<T>::getPixelCode() const’ [-fpermissive]
     virtual int getPixelCode() const override;
                 ^
/home/traversaro/src/something-superbuild/something/YARP/src/libYARP_sig/include/yarp/sig/Image.h:623:69: error: specialization of ‘template<class T> virtual int yarp::sig::ImageOf<T>::getPixelCode() const’ in different namespace [-fpermissive]
 inline int yarp::sig::ImageOf<yarp::sig::PixelBgra>::getPixelCode() const {
                                                                     ^
/home/traversaro/src/something-superbuild/something/YARP/src/libYARP_sig/include/yarp/sig/Image.h:563:17: error:   from definition of ‘template<class T> virtual int yarp::sig::ImageOf<T>::getPixelCode() const’ [-fpermissive]
     virtual int getPixelCode() const override;
                 ^
/home/traversaro/src/something-superbuild/something/YARP/src/libYARP_sig/include/yarp/sig/Image.h:628:75: error: specialization of ‘template<class T> virtual int yarp::sig::ImageOf<T>::getPixelCode() const’ in different namespace [-fpermissive]
 inline int yarp::sig::ImageOf<yarp::sig::PixelMonoSigned>::getPixelCode() const {
                                                                           ^
/home/traversaro/src/something-superbuild/something/YARP/src/libYARP_sig/include/yarp/sig/Image.h:563:17: error:   from definition of ‘template<class T> virtual int yarp::sig::ImageOf<T>::getPixelCode() const’ [-fpermissive]
     virtual int getPixelCode() const override;
                 ^
/home/traversaro/src/something-superbuild/something/YARP/src/libYARP_sig/include/yarp/sig/Image.h:633:74: error: specialization of ‘template<class T> virtual int yarp::sig::ImageOf<T>::getPixelCode() const’ in different namespace [-fpermissive]
 inline int yarp::sig::ImageOf<yarp::sig::PixelRgbSigned>::getPixelCode() const {
                                                                          ^
/home/traversaro/src/something-superbuild/something/YARP/src/libYARP_sig/include/yarp/sig/Image.h:563:17: error:   from definition of ‘template<class T> virtual int yarp::sig::ImageOf<T>::getPixelCode() const’ [-fpermissive]
     virtual int getPixelCode() const override;
                 ^
/home/traversaro/src/something-superbuild/something/YARP/src/libYARP_sig/include/yarp/sig/Image.h:638:70: error: specialization of ‘template<class T> virtual int yarp::sig::ImageOf<T>::getPixelCode() const’ in different namespace [-fpermissive]
 inline int yarp::sig::ImageOf<yarp::sig::PixelFloat>::getPixelCode() const {
                                                                      ^
/home/traversaro/src/something-superbuild/something/YARP/src/libYARP_sig/include/yarp/sig/Image.h:563:17: error:   from definition of ‘template<class T> virtual int yarp::sig::ImageOf<T>::getPixelCode() const’ [-fpermissive]
     virtual int getPixelCode() const override;
                 ^
/home/traversaro/src/something-superbuild/something/YARP/src/libYARP_sig/include/yarp/sig/Image.h:643:73: error: specialization of ‘template<class T> virtual int yarp::sig::ImageOf<T>::getPixelCode() const’ in different namespace [-fpermissive]
 inline int yarp::sig::ImageOf<yarp::sig::PixelRgbFloat>::getPixelCode() const {
                                                                         ^
/home/traversaro/src/something-superbuild/something/YARP/src/libYARP_sig/include/yarp/sig/Image.h:563:17: error:   from definition of ‘template<class T> virtual int yarp::sig::ImageOf<T>::getPixelCode() const’ [-fpermissive]
     virtual int getPixelCode() const override;
                 ^
/home/traversaro/src/something-superbuild/something/YARP/src/libYARP_sig/include/yarp/sig/Image.h:648:71: error: specialization of ‘template<class T> virtual int yarp::sig::ImageOf<T>::getPixelCode() const’ in different namespace [-fpermissive]
 inline int yarp::sig::ImageOf<yarp::sig::PixelRgbInt>::getPixelCode() const {
                                                                       ^
/home/traversaro/src/something-superbuild/something/YARP/src/libYARP_sig/include/yarp/sig/Image.h:563:17: error:   from definition of ‘template<class T> virtual int yarp::sig::ImageOf<T>::getPixelCode() const’ [-fpermissive]
     virtual int getPixelCode() const override;
[... cut ...]
     virtual int getPixelCode() const override;
                 ^
src/libYARP_sig/CMakeFiles/YARP_sig.dir/build.make:62: recipe for target 'src/libYARP_sig/CMakeFiles/YARP_sig.dir/src/ImageCopy.cpp.o' failed
make[2]: *** [src/libYARP_sig/CMakeFiles/YARP_sig.dir/src/ImageCopy.cpp.o] Error 1
CMakeFiles/Makefile2:539: recipe for target 'src/libYARP_sig/CMakeFiles/YARP_sig.dir/all' failed
make[1]: *** [src/libYARP_sig/CMakeFiles/YARP_sig.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2
~~~

traversaro, 2017-10-27:
Command to reproduce the problem (with a proper value for `YARP_BUILD_DIRECTORY` and  `YARP_SOURCE_DIRECTORY` : 
~~~
cd ${YARP_BUILD_DIRECTORY}/src/libYARP_sig && /usr/bin/g++  -DYARP_PRESENT -DYARP_sig_EXPORTS -D_REENTRANT -I${YARP_BUILD_DIRECTORY}/src/libYARP_conf/include -I${YARP_SOURCE_DIRECTORY}/src/libYARP_OS/include -I${YARP_SOURCE_DIRECTORY}/src/libYARP_sig/include  -Wall -Wextra -Wsign-compare -Wpointer-arith -Winit-self -Wnon-virtual-dtor -Wcast-align -Wunused -Wunused-but-set-variable -Wvla -Wmissing-include-dirs -Wlogical-op -Wreorder -Wsizeof-pointer-memaccess -Woverloaded-virtual -Wundef -Wredundant-decls -Wunknown-pragmas -Wunused-result -Wc++11-compat -Wsuggest-override  -Wno-unused-parameter  -fvisibility=hidden -fvisibility-inlines-hidden  -Wdeprecated-declarations -O3 -DNDEBUG -fPIC   -std=c++11 -o CMakeFiles/YARP_sig.dir/src/ImageCopy.cpp.o -c ${YARP_SOURCE_DIRECTORY}/src/libYARP_sig/src/ImageCopy.cpp
~~~
The same command with `clang++` seems to work.

traversaro, 2017-10-27:
CMakeCache.txt : [CMakeCache.txt](https://github.com/robotology/yarp/files/1422308/CMakeCache.txt)
.

traversaro, 2017-10-27:
Reproduced on both gcc `Ubuntu 5.4.0-6ubuntu1~16.04.4` and `Ubuntu 5.4.0-6ubuntu1~16.04.5`.

traversaro, 2017-10-27:
The problem was introduced by commit b6426055e7f458981a114985d8aad7e579da3d46 .

traversaro, 2017-10-27:
Unable to reproduce on a default build.  : (

drdanz, 2017-10-27:
Weird, it works on all the other compilers tested... Can you try adding

```c++
namespace yarp {
namespace sig {
```

To this part and remove it from the declarations, and see if it fixes the problem?

traversaro, 2017-10-27:
@drdanz That exactly the solution that I committed in https://github.com/robotology/yarp/pull/1434 , and is working fine for me. Were you able to reproduce the issue on a Ubuntu 16.04 setup?

francesco-romano, 2017-10-27:
This happens also on the robot setup (@S-Dafarra)

gabrielenava, 2017-10-29:
Same happened to me today while trying to compile after an update, on Ubuntu 16.04.

aerydna, 2017-11-02:
can we close this issue? @traversaro @drdanz

aerydna, 2017-11-02:
i'll close it. if it is a mistake reopen it..

closed 2017-11-02 by aerydna

traversaro, 2017-11-02:
Yep the issue was solved.