robotology/yarp #1437: Never returning call to Time::useNetworkClock due to the use of persistent connection

traversaro, 2017-10-31:
I had the impression that something was wrong in the `yarp::os::NetworkClock` behaviour since https://github.com/robotology/yarp/pull/1277 (see https://github.com/robotology/yarp/commit/c574b3c371371c6bb9a14e9afdfbcad8245c0d18#commitcomment-24627511 ), but I was never able to reproduce the problem. 

I was able now to reproduce the problem in failing test case in YARP, added in PR . 
In a nutshell, even if the port `/clock` is up and working fine, the NetworkClock that is reading from `/clock` is never valid, and hence any call to `useNetworkClock("clock")` blocks and never returns (see https://github.com/robotology/yarp/issues/1377). Switching back to not using a persistent connection in https://github.com/robotology/yarp/commit/c574b3c371371c6bb9a14e9afdfbcad8245c0d18#diff-174cf88b44c18517c70a685e17736365R65 and https://github.com/robotology/yarp/commit/c574b3c371371c6bb9a14e9afdfbcad8245c0d18#diff-174cf88b44c18517c70a685e17736365R53) apparently solves the problem. 

cc @barbalberto

barbalberto, 2017-10-31:
Thanks for reporting, I'll check. 
Do you think the root of the issue is the persistant connection?

traversaro, 2017-10-31:
I tried by removing the persistent connection in https://github.com/robotology/yarp/commit/c574b3c371371c6bb9a14e9afdfbcad8245c0d18#diff-174cf88b44c18517c70a685e17736365R65 and https://github.com/robotology/yarp/commit/c574b3c371371c6bb9a14e9afdfbcad8245c0d18#diff-174cf88b44c18517c70a685e17736365R53 and the test is successful.

traversaro, 2017-11-01:
If using an external NetworkClock (i.e. not using useNetworkClock) a possible workaround is to force the (non-persistent) connection, as done in https://github.com/robotology/yarp/pull/1438/files#diff-3e0a9b8a09d2e1a159e06f572cdfdd55R82 .

traversaro, 2017-11-20:
If we don't have the time to tackle this issue, I propose that for the time being we switch to use a non-persistent connection for the clock by default. 
As this change was only introduced on `devel`, for sure no user is relying on this connection being persistent.

traversaro, 2017-11-20:
cc @drdanz

lornat75, 2017-11-22:
@barbalberto I agree with @traversaro

barbalberto, 2017-11-22:
:+1: 
Let's remove the default persistancy now. It was a nice behaviour, but the network clock works also without it.
Then we can use the test to debug persistancy and, in case, we can reintroduce it later.

barbalberto, 2017-11-22:
I tried the test and I found out that:

_Using CTest with:_ "make test"
**The test always fails**, both with persistant and normal connections.

_Manually calling RTF_ with:   testrunner -t  /.../NetworkClockTest.so
**The test always succeed**, both with persistant or normal connections.

So, if a problem do exist with persistancy, I think this test is not actually catching it. Instead it shows to me a strange behaviour of rtf when called by ctest. (Maybe the fixture is not created in time for the test to be executed.)

However, since persistancy is not mandatory, I made the requested change anyway in commit efe1b2b5faaeb153c362e4cba70c9da35cabca2d

traversaro, 2017-12-12:
>  The test always fails, both with persistant and normal connections. 

Interesting. Also after explicit non-persistent connection ( https://github.com/robotology/yarp/pull/1438/files#diff-3e0a9b8a09d2e1a159e06f572cdfdd55R82 ) the clock is still not valid?