robotology/yarp #1445: yarpmotorgui does not load the parts specified in the .ini file

plinioMoreno, 2017-11-02:
While running the yarpmotorgui using the config file [vizzy_sim_motor_gui.ini](https://github.com/vislab-tecnico-lisboa/vizzy/blob/master/vizzy_yarp_icub/app/vizzyRobotMotorGui/vizzy_sim_motor_gui.ini), I get the following output

```
yarpmotorgui --from ./vizzy_sim_motor_gui.ini
||| configuring
||| default config file specified as ./vizzy_sim_motor_gui.ini
||| checking [/home/plinio/repositories/vizzy_ws/src/vizzy/vizzy_yarp_icub/app/vizzyRobotMotorGui/./vizzy_sim_motor_gui.ini] (pwd)
||| found /home/plinio/repositories/vizzy_ws/src/vizzy/vizzy_yarp_icub/app/vizzyRobotMotorGui/./vizzy_sim_motor_gui.ini
QObject::connect: No such signal MainWindow::internalClose()
QObject::connect:  (sender name:   'MainWindow')
QObject::connect:  (receiver name: 'MainWindow')
[WARNING]robotDescriptionServer not found, robot parts will be set manually. 
||| finding paths [plugins]
||| checking [/home/plinio/repositories/vizzy_ws/src/vizzy/vizzy_yarp_icub/app/vizzyRobotMotorGui/plugins] (pwd)
||| checking [/home/plinio/repositories/vizzy_ws/src/vizzy/vizzy_yarp_icub/app/vizzyRobotMotorGui/./plugins] (defaultConfigFile path)
||| checking [/home/plinio/.config/yarp/plugins] (YARP_CONFIG_HOME)
||| checking [/home/plinio/.local/share/yarp/plugins] (YARP_DATA_HOME)
||| checking [/etc/xdg/xdg-ubuntu/yarp/plugins] (YARP_CONFIG_DIRS)
||| checking [/usr/share/upstart/xdg/yarp/plugins] (YARP_CONFIG_DIRS)
||| checking [/etc/xdg/yarp/plugins] (YARP_CONFIG_DIRS)
||| checking [/home/plinio/repositories/yarp/build/share/yarp/plugins] (YARP_DATA_DIRS)
||| found /home/plinio/repositories/yarp/build/share/yarp/plugins
||| checking [/home/plinio/repositories/icub-main/build/share/iCub/plugins] (YARP_DATA_DIRS)
yarp: Port /log/deepthought/yarpmotorgui/6429 active at tcp://10.0.27.23:10021/
LogForwarder error while connecting port /log/deepthought/yarpmotorgui/6429
[DEBUG]Appending /icub/head
[DEBUG]Appending /icub/torso
[DEBUG]Appending /icub/left_arm
[DEBUG]Appending /icub/right_arm
[DEBUG]Appending /icub/left_leg
[DEBUG]Appending /icub/right_leg```

And the GUI window shows only icub parts. Am I missing another configuration file(s)?
How should I proceed to remove the warning about the missing ```robotDescriptionServer```

plinioMoreno, 2017-11-03:
The error was also happening on the real robot, so with the help of @gsaponaro we added the robotDescription.xml in [aebbc16](https://github.com/robotology/robots-configuration/commit/aebbc169ae376b244c03f479dd8ef5047554dfd5) and now yarpmotorgui on the real Vizzy is working.

But I still keep the issue open, because the correct message when the robotDescriptionServer must be ```[ERROR]``` and not ```[WARNING]```

jgvictores, 2017-11-05:
On TEO we had issues (was defaulting to `icub` rather than [`teo`](http://roboticslab.uc3m.es/roboticslab/robot/teo-humanoid)) until we changed `name` to `robot` at [this commit](https://github.com/roboticslab-uc3m/teo-configuration-files/commit/78f0e458065f7f87fe94720cea584d857fa8a6f5), which references [this YARP commit](https://github.com/robotology/yarp/commit/da5cef74557ab1c610c6f386ca564e3421d2a080).

Yes, more documentation on `robotDescription ` would be great!!

randaz81, 2017-11-06:
robotDescription is optional and still work in progress. It allows yarpmotorgui and other modules to automatically detect the name and the parts of the robot which are currently running.

Until the next release, I suggest you to continue to use the `yarpmotorgui.ini `file as usual. (`name` parameter has changed meaning long time ago: it is used to identify the full name of the ports e.g. /icub/left_arm.  You need to use `robot` e.g. icub and `parts `e.g. left_arm).

randaz81, 2017-11-06:
moved to #1447

closed 2017-11-06 by randaz81