robotology/yarp #1450: Travis random failing not due to code

barbalberto, 2017-11-08:
Often Travis CI fails by his own problems, resulting in a errored PR.
Relaunching the job sometime make it work, sometime the error changes.

Hereafter are errors of the same PR, different jobs
```
1) /Users/travis/.travis/job_stages: line 57: shell_session_update: command not found
Done. Your build exited with 1.
/Users/travis/.travis/job_stages: line 166: shell_session_update: command not found
-----------------------------------------
2) E: Failed to fetch http://ppa.launchpad.net/ubuntu-toolchain-r/test/ubuntu/pool/main/g/gcc-4.8/libgfortran-4.8-dev_4.8.5-2ubuntu1~14.04.1_amd64.deb  Unable to connect to ppa.launchpad.net:http:
E: Failed to fetch http://ppa.launchpad.net/ubuntu-toolchain-r/test/ubuntu/pool/main/g/gcc-4.8/gfortran-4.8_4.8.5-2ubuntu1~14.04.1_amd64.deb  Unable to connect to ppa.launchpad.net:http:
```

drdanz, 2017-11-08:
This is unfortunately a know issue, but there is nothing we can do about...

closed 2017-11-08 by drdanz

traversaro, 2017-11-30:
The frequency of this spurious failure seems to be increasing, especially on macOS builds.

traversaro, 2017-11-30:
cc @nunoguedelha