robotology/yarp #1461: yarpserver crashes using persistent connections

claudiofantacci, 2017-11-22:
Following issue #1437 and PR #1460, I would like to report the following issue involving `persistent connections` and `yarpserver`.

I used to use persistent connection in one of my YARP module in order to avoid reconnecting ports every time the module was rerun.
I noticed that after several runs, between 8 and 10, of my module `yarpserver` was crashing.
By removing persistent connection crashes disappeared.

Please consider the following:
 - this issue happened quite some months ago, an I lost track of it, my bad 😔
 - I don't recall which was exactly the error upon `yarpserver` crash, it may was a segmentation fault
 - the issue was quite repeatable and I am sure that the application was always closing correctly

claudiofantacci, 2017-11-22:
cc @traversaro @barbalberto @drdanz @lornat75

barbalberto, 2017-11-22:
Thanks for reporting, if you remember more details and/or have time to write a test case that'll be awesome!

claudiofantacci, 2017-11-22:
@barbalberto it should not be difficult, the application should be like this:
 1. Module A opens some connections
 2. Module B opens some connections
 3. Connect them with persistent option
 4. Send some data
 5. Close the applications smoothly
 6. Repeat from 1. several times
which is a reasonable approximation of what was done in my module.

Having said that, I can write a test bench as soon as I have some time 👍

closed 2020-12-22 by drdanz