robotology/yarp #1464: yarp devel does not compile on macOS Sierra with CMake 3.10.0

traversaro, 2017-11-23:
Probably this is related to some AUTOMOC changes in the latest cmake: 
~~~
CMake Warning (dev) in src/yarpview/plugin/CMakeLists.txt:
  Policy CMP0071 is not set: Let AUTOMOC and AUTOUIC process GENERATED files.
  Run "cmake --help-policy CMP0071" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  For compatibility, CMake is excluding the GENERATED source file(s):

    "/Users/traversaro/src/yarp/build/src/yarpview/plugin/qrc_res.cpp"

  from processing by AUTOMOC.  If any of the files should be processed, set
  CMP0071 to NEW.  If any of the files should not be processed, explicitly
  exclude them by setting the source file property SKIP_AUTOMOC:

    set_property(SOURCE file.h PROPERTY SKIP_AUTOMOC ON)

This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) in src/yarpview/src/CMakeLists.txt:
  Policy CMP0071 is not set: Let AUTOMOC and AUTOUIC process GENERATED files.
  Run "cmake --help-policy CMP0071" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  For compatibility, CMake is excluding the GENERATED source file(s):

    "/Users/traversaro/src/yarp/build/src/yarpview/src/qrc_res.cpp"

  from processing by AUTOMOC.  If any of the files should be processed, set
  CMP0071 to NEW.  If any of the files should not be processed, explicitly
  exclude them by setting the source file property SKIP_AUTOMOC:

    set_property(SOURCE file.h PROPERTY SKIP_AUTOMOC ON)

This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) in src/yarpview/src/CMakeLists.txt:
  Policy CMP0071 is not set: Let AUTOMOC and AUTOUIC process GENERATED files.
  Run "cmake --help-policy CMP0071" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  For compatibility, CMake is excluding the GENERATED source file(s):

    "/Users/traversaro/src/yarp/build/src/yarpview/src/qrc_res.cpp"

  from processing by AUTOMOC.  If any of the files should be processed, set
  CMP0071 to NEW.  If any of the files should not be processed, explicitly
  exclude them by setting the source file property SKIP_AUTOMOC:

    set_property(SOURCE file.h PROPERTY SKIP_AUTOMOC ON)

This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) in src/yarpscope/plugin/CMakeLists.txt:
  Policy CMP0071 is not set: Let AUTOMOC and AUTOUIC process GENERATED files.
  Run "cmake --help-policy CMP0071" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  For compatibility, CMake is excluding the GENERATED source file(s):

    "/Users/traversaro/src/yarp/build/src/yarpscope/plugin/qrc_res.cpp"

  from processing by AUTOMOC.  If any of the files should be processed, set
  CMP0071 to NEW.  If any of the files should not be processed, explicitly
  exclude them by setting the source file property SKIP_AUTOMOC:

    set_property(SOURCE file.h PROPERTY SKIP_AUTOMOC ON)

This warning is for project developers.  Use -Wno-dev to suppress it.

[... 239 lines cut ...]
  Cannot find source file:

    /Users/traversaro/src/yarp/src/yarplogger/advanced_dialog.cpp;/Users/traversaro/src/yarp/src/yarplogger/logtab.cpp;/Users/traversaro/src/yarp/src/yarplogger/logtabSorting.cpp;/Users/traversaro/src/yarp/src/yarplogger/main.cpp;/Users/traversaro/src/yarp/src/yarplogger/mainwindow.cpp;/Users/traversaro/src/yarp/src/yarplogger/messageWidget.cpp;/Users/traversaro/src/yarp/src/yarplogger/yarprunPortSorting.cpp;/Users/traversaro/src/yarp/src/yarplogger/advanced_dialog.h;/Users/traversaro/src/yarp/src/yarplogger/logtab.h;/Users/traversaro/src/yarp/src/yarplogger/logtabSorting.h;/Users/traversaro/src/yarp/src/yarplogger/mainwindow.h;/Users/traversaro/src/yarp/src/yarplogger/messageWidget.h;/Users/traversaro/src/yarp/src/yarplogger/yarprunPortSorting.h;/Users/traversaro/src/yarp/src/yarplogger/res.qrc;/Users/traversaro/src/yarp/build/src/yarplogger/qrc_res.cpp;/Users/traversaro/src/yarp/src/yarplogger/advanced_dialog.ui;/Users/traversaro/src/yarp/src/yarplogger/logtab.ui;/Users/traversaro/src/yarp/src/yarplogger/mainwindow.ui;/Users/traversaro/src/yarp/build/src/yarplogger/ui_advanced_dialog.h;/Users/traversaro/src/yarp/build/src/yarplogger/ui_logtab.h;/Users/traversaro/src/yarp/build/src/yarplogger/ui_mainwindow.h

  Tried extensions .c .C .c++ .cc .cpp .cxx .m .M .mm .h .hh .h++ .hm .hpp
  .hxx .in .txx

CMake Error in src/yarpdataplayer/CMakeLists.txt:
  Cannot find source file:

    src/aboutdlg.cpp;src/genericinfodlg.cpp;src/loadingwidget.cpp;src/main.cpp;src/mainwindow.cpp;src/utils.cpp;src/worker.cpp;include/aboutdlg.h;include/genericinfodlg.h;include/loadingwidget.h;include/log.h;include/mainwindow.h;include/utils.h;include/worker.h;src/RC/res.qrc;/Users/traversaro/src/yarp/build/src/yarpdataplayer/qrc_res.cpp;include/UI/aboutdlg.ui;include/UI/genericinfodlg.ui;include/UI/loadingwidget.ui;include/UI/mainwindow.ui;/Users/traversaro/src/yarp/build/src/yarpdataplayer/ui_aboutdlg.h;/Users/traversaro/src/yarp/build/src/yarpdataplayer/ui_genericinfodlg.h;/Users/traversaro/src/yarp/build/src/yarpdataplayer/ui_loadingwidget.h;/Users/traversaro/src/yarp/build/src/yarpdataplayer/ui_mainwindow.h;/Users/traversaro/src/yarp/src/yarpdataplayer/idl_generated_code/src/yarpdataplayer_IDL.cpp;/Users/traversaro/src/yarp/src/yarpdataplayer/idl_generated_code/include/yarpdataplayer_IDL.h;yarpdataplayer.thrift

  Tried extensions .c .C .c++ .cc .cpp .cxx .m .M .mm .h .hh .h++ .hm .hpp
  .hxx .in .txx

CMake Error in src/yarpmotorgui/CMakeLists.txt:
  Cannot find source file:

    main.cpp;flowlayout.cpp;jointitem.cpp;mainwindow.cpp;partitem.cpp;piddlg.cpp;sequencewindow.cpp;sliderOptions.cpp;sliderWithTarget.cpp;startdlg.cpp;flowlayout.h;jointitem.h;log.h;mainwindow.h;partitem.h;piddlg.h;sliderOptions.h;sliderWithTarget.h;yarpmotorgui.h;sequencewindow.h;startdlg.h;res.qrc;/Users/traversaro/src/yarp/build/src/yarpmotorgui/qrc_res.cpp;jointitem.ui;mainwindow.ui;piddlg.ui;sequencewindow.ui;startdlg.ui;sliderOptions.ui;/Users/traversaro/src/yarp/build/src/yarpmotorgui/ui_jointitem.h;/Users/traversaro/src/yarp/build/src/yarpmotorgui/ui_mainwindow.h;/Users/traversaro/src/yarp/build/src/yarpmotorgui/ui_piddlg.h;/Users/traversaro/src/yarp/build/src/yarpmotorgui/ui_sequencewindow.h;/Users/traversaro/src/yarp/build/src/yarpmotorgui/ui_startdlg.h;/Users/traversaro/src/yarp/build/src/yarpmotorgui/ui_sliderOptions.h

  Tried extensions .c .C .c++ .cc .cpp .cxx .m .M .mm .h .hh .h++ .hm .hpp
  .hxx .in .txx

-- Generating done
~~~

traversaro, 2017-11-23:
Even by modifying the yarpmotorgui cmake line to:
~~~
 add_executable(yarpmotorgui  main.cpp flowlayout.cpp)
~~~
I still get the error: 
~~~
CMake Error in src/yarpmotorgui/CMakeLists.txt:
  Cannot find source file:

    main.cpp;flowlayout.cpp

  Tried extensions .c .C .c++ .cc .cpp .cxx .m .M .mm .h .hh .h++ .hm .hpp
  .hxx .in .txx
~~~

claudiofantacci, 2017-11-23:
As discussed f2f, this problem my be related to CMake 3.10. I have version 3.9.4 and I don't have this issue.

traversaro, 2017-11-23:
The problem is in:
~~~cmake
yarp_osx_duplicate_and_add_bundle(TARGET yarpmotorgui
                                   INSTALL_COMPONENT utilities
                                 INSTALL_DESTINATION "${CMAKE_INSTALL_BINDIR}")
~~~

Commenting out this call the problem disappears.

traversaro, 2017-11-23:
Discussing with @drdanz it seems it could be a CMake regression.

closed 2017-11-27 by drdanz