robotology/yarp #1470: Aborted core opening ports starting with "="

Nicogene, 2017-11-28:
For unknown reasons it is supported `=`as starting character for port names, in alternative to `/` but it crashes 😄 e.g.:
```
yarp read =ciao
terminate called after throwing an instance of 'std::out_of_range'
  what():  basic_string::substr: __pos (which is 18446744073709551615) > this->size() (which is 4)
Aborted (core dumped)
```
The magic number `18446744073709551615` should be `std::string::npos`.

To possible fixes:

- We drop the `=` as starting character( there will be someone that used to open ports with `=`). 👍 
- We fix this bug. ❤️ 

Vote what you prefer!

Yet Another Mysterious Behaviour Supported.

drdanz, 2017-12-01:
I'd say we drop the '=' starting character, but we should definitely investigate whether there is some reason why this was added.

Nicogene, 2017-12-19:
@damn1 and I discovered that adding in `NestedContact.cpp`
```
        if (idx == ConstString::npos) return false;
```
for avoing this out of range, typing `yarp read =ciao` you will have:
```
yarp: Port /tmp/port/1 active at tcp://10.255.111.75:10002/
yarp: Port /tmp/port/2 active at tcp://10.255.111.75:10004/
yarp: Port /tmp/port/2=ciao active at tcp://10.255.111.75:10003/
```
So we tried, in `Port.cpp`, to remove `=` among the allowed starting character and we had:
```
yarp: Port /tmp/port/1 active at tcp://10.255.111.75:10002/
yarp: Port name '=ciao/tmp/port/1' needs to start with a '/' character
```
without closing the yarp read...

On the other hand we found a "working" usage of `=`
```
yarp read /ciao=/ciaone
yarp: Port /ciao active at tcp://10.255.111.75:10005/
yarp: Port /ciaone-@/ciao active at tcp://10.255.111.75:10004/

```

damn1, 2017-12-27:
| **@** |   | **=**
-- | ---- | -- | ------
`/p1@/p2` | - node `/p2` is opened<br>- open port `/p1[cat]@/p2`<br>- subscribe to topic `/p1` | `/p2=/p1` | same result
`/p1@p2` | - :exclamation: no leading "/" for `p2`<br>- open port `/p1[cat]@p2`<br>- subscribe to topic `/p1` | `p2=/p1` | `p2=/p1` has to start with “/”, do nothing
`p1@/p2` | no leading “/”, do nothing | `/p2=p1` | :exclamation: DUMP
`/p1@` | - node /tmp/port/(n)<br>- open `/p1[cat]@/tmp/port/(n)`<br>- subscribe to topic `/p1` | `=/p1` | - node `/tmp/port/(n)`<br>- :exclamation: open `/p1/tmp/port/n[cat]@`<br>- :exclamation: subscribe to topic `/p1/tmp/port/(n)`<br>(with a ‘node name mismatch’ fail)
`p1@` | no leading “/”, do nothing | `=p1` |  :exclamation: DUMP
`p1/p2@` | no leading “/”, do nothing | `=p1/p2` | - :exclamation: `p1` is taken as the category<br>- node `tmp/port/(n)` opened<br>- :exclamation: also node `tmp/port/(n+1)` is opened<br>- :exclamation: also `/tmp/port/(n+1)=p1/p2` is opened.<br>No topic subscription.
`@/p1` | no leading “/”, do nothing | `/p1=` |  :exclamation: DUMP
`@` | no leading “/”, do nothing | `=` |  :exclamation: DUMP
`/p1@/` |- :exclamation: node `/` is opened<br>- open `/p1[cat]@/`<br>- subscribe to `/p1` | `/=/p1` | same result

This table shows the result of the use of the two syntaxes. I added an :exclamation: where I think the behaviour is broken. So if its ok, I would drop the `=` checks as @drdanz commented. And I would add a regex control function as first thing to be called on the input string in `Port::open()`, to avoid problems I found in the `@` syntax, like registering a port named just `/`.

damn1, 2017-12-29:
Some other information:
1. also `/node#/port` syntax is supported as alternative to '@' syntax.
2. registering ports named `...string` is allowed, but then they are not connectable since #1441
3. `yarp read ""` results in `yarp: Anonymous port active at tcp://10.255.111.155:37571/` (port number is quite random) but this port is not registered and is not visible in the name server.

Since I'm already writing a patch I would drop 2. and 3. as I think they're not expected behaviours.

damn1, 2018-01-04:
Another buggy behaviour:
```
yarp read /n1=-/p
yarp: Port /n1 active at tcp://10.255.111.155:10006/
yarp: Port /n1=-/p active at tcp://10.255.111.155:10005/
```

```
yarp read /n=/p
yarp: Port /n active at tcp://10.255.111.155:10003/
yarp: Port /p-@/n active at tcp://10.255.111.155:10002/
```
If you explicit category, yarp leave the port name as it is. If you don't specifify it yarp chooses one and changes the name syntax from '=' to '@' .
I think the behaviour should be unified, change always or change never. Same happens with '#'.
Moreover, if you specify category, there is no check on `currentCore->commitToRead` or `currentCore->commitToWrite` so one can open a port with `+` category with `read` command...

Also I see from `NodeTest` before testing names, this:
    `report(0,"check support for various experimental node/topic/service schemes");`
So if these syntaxes are 'experimental' we could unify them to '@' only.