robotology/yarp #1474: YARP-ROS communication fails with ROS running on WSL

randaz81, 2017-11-28:
Tested on Windows 10, build 16299 (Fall Creators Update). 
WSL Running Ubuntu 16.04.03 with ROS lunar.

ROS fails to perform connections due to a suspected bug in **ros_comm** `1.13.2-0xenial-20171012-115727-0800`. A simple test case is rviz failing to display any /tf, published either by ROS or YARP. 
Requires more investigation on next ROS release.

randaz81, 2017-11-28:
Relevant infos:

- https://answers.ros.org/question/238646/installing-ros-on-ubuntu-bash-in-windows-10/
- https://github.com/ros/ros_comm/pull/1050
- https://github.com/ros/ros_comm/pull/1202

drdanz, 2017-12-01:
This does not look like a bug in YARP...

randaz81, 2017-12-03:
Indeed. It is not. I put the issue here so if someone searches WSL in the YARP issues can know that someone has already tested it and that some problems have been found.

drdanz, 2017-12-04:
Closing then... 😉

closed 2017-12-04 by drdanz