robotology/yarp #1476: Some plugins are only compiled if ACE is used, even if apparently they do not depend on ACE

traversaro, 2017-11-30:
See: 
* https://github.com/robotology/yarp/search?utf8=%E2%9C%93&q=DEPENDS+%22CREATE_OPTIONAL_CARRIERS%3BYARP_HAS_ACE%22&type=
* https://github.com/robotology/yarp/blob/9a395f6e41e01d8d350b6a4e5e2b2f6c6f913b74/src/devices/serial/CMakeLists.txt#L10

I tried a few of them, and they could be compiled even if ACE was not present in the system.

drdanz, 2017-12-01:
I doubt this would compile without ACE:

https://github.com/robotology/yarp/blob/9a395f6e41e01d8d350b6a4e5e2b2f6c6f913b74/src/devices/serial/SerialDeviceDriver.h#L16-L17

traversaro, 2017-12-01:
Indeed, "a few of them" referenced to the fact that I did not try the serial device. All other plugins (carriers) listed in https://github.com/robotology/yarp/search?utf8=%E2%9C%93&q=DEPENDS+%22CREATE_OPTIONAL_CARRIERS%3BYARP_HAS_ACE%22&type= compile fine even without ACE installed in the system.

drdanz, 2017-12-01:
I see, I think that this might be some leftover, since the `Carriers` class was in impl, forcing carriers to use the impl library, and probably ACE as well.

I think that these carriers should be working without ace:

* portmonitor_carrier
* tcpros_carrier
* xmlrpc_carrier

PR #1477 (2017-12-04): Remove unused ACE dependencies on carriers
Fixes #1476
commit 6ce34f00 (2017-12-04): Remove unused ACE dependencies on carriers

Fixes #1476
closed 2017-12-04 by drdanz