robotology/yarp #1478: Flaky math::RandTest failure in AppVeyor

traversaro, 2017-12-01:
https://ci.appveyor.com/project/robotology/yarp/build/appveyor.3714/job/qqxvwhudl3a29eei

~~~
56/84 Test #56: math::RandTest .....................***Failed    0.06 sec
  yarp: YARP_VERBOSE environment variable is set
  yarp: 0 | RandTest: checking random  number generation...
  yarp: 0 | RandTest:   [sequence is <= 1] passed ok
  yarp: 0 | RandTest:   [sequence is >= 0] passed ok
  yarp: 0 | RandTest:   [average is ~0.5] passed ok
  yarp: 0 | RandTest:   [std is ~0.28] passed ok
  yarp: 0 | RandTest:   [same seed produces identical sequences] passed ok
  yarp: 0 | RandTest:   [different seeds generate different sequences] passed ok
  yarp: 0 | RandTest:   [default seed initialization for two sequences] passed ok
  yarp: 0 | RandTest: checking random  matrix generation...
  yarp: 0 | RandTest:   [random matrix] passed ok
  yarp: 0 | RandTest: checking random generation with normal distribution...
  yarp: 1 | RandTest:   FAILURE in file C:\projects\yarp\tests\libYARP_math\RandTest.cpp:79 [normal distribution average ~as requested] avOk (0) == true (1)
  yarp: 0 | RandTest:   [normal distribution std ~as requested] passed ok
  yarp: 0 | RandTest: A PROBLEM WAS ENCOUNTERED
  yarp: 0 | A PROBLEM WAS ENCOUNTERED
  Going to generate 10000 numbers, normally distributed in the range [0-1]...done!
  Now performing some *basic* tests on the sequence
  Maximum value was: 0.999990
  Minimum value was: 0.000106
  Average was: 0.494474, std was: 0.288404
  Going to generated 10000 samples with u=1.000000 and sigma=2.000000...done!
  Now performing some *basic* tests on the sequence
  Average: 1.050629
  Std: 1.995386
~~~

traversaro, 2017-12-01:
cc @damn1 
If you want to debug this, ctest options `--output-on-failure` and `--repeat-until-fail <n>` are your great friends.

drdanz, 2017-12-01:
I'd say that it's kind of normal for a test on random number to randomly fail, otherwise it would not be random!
Probably, `fabs(average-u)<0.05` over 10000 samples is not a good assumption

damn1, 2017-12-01:
Yes the problem is actually the order of the numbers. I can fix this increasing that `0.05` if you agree.

drdanz, 2017-12-04:
@damn1 Did you do a few tests on what would be a good number? I've been running the test on my system with `--output-on-failure --repeat-until-fail 10000` and I did not have a single failure, rerun with `100000` and I finally had a failure... It's weird that on appveyor this happened a few times already, perhaps windows implementation is weaker?

Anyway I modified it to print the actual result, and after running it a few million times

```
fabs(average-u) = 0.054968
fabs(average-u) = 0.067257
fabs(average-u) = 0.055323
fabs(average-u) = 0.053686
fabs(average-u) = 0.064522
fabs(average-u) = 0.051500
fabs(average-u) = 0.060231
```

I think `0.075` might be a good value, @traversaro what do you think?

traversaro, 2017-12-05:
If I understood correctly the semantics of the sigma parameter (that is not documented properly, but I guess it is the standard deviation of the Gaussian distribution), the standard deviation of the tested distribution is `σ = 2.0` . The standard deviation of the outcome of this distribution average `N = 10000` times is `σ̅= sqrt((σ^2)/N) = 0.02`.

traversaro, 2017-12-05:
If the standard deviation of the average is `0.02`, then taking a threshold of `0.05 = 0.02*2.5` should give a 98.7% probability that a test passes, i.e. 1 out of 81 tests will fail (see https://en.wikipedia.org/wiki/68%E2%80%9395%E2%80%9399.7_rule). This does not seem consistent with our tests, and this is strange (or perhaps I did something wrong in the standard deviations computations). Based on this computations, a threshold of `0.02 * 7 = 0.14` will fail only once in 390682215445 test, that I think is a good tradeoff.

drdanz, 2017-12-19:
There is a big problem with tests on random number. In order to be *random* the test **must** fail from time to time.
For now I increased these thresholds to 0.75 in 
e85dd28, nonetheless I've seen failures in other places in the file, so we might have to reopen this issue in the future

commit e85dd28c (2017-12-19): Tests: Increase acceptable average for math test

Fixes #1478
closed 2017-12-19 by drdanz